aircrack-ng/rtl8812au #263: Not compiling since last batch of commits linux 4.19.17

wifislax-ng, 2019-01-25:
SInce last batch of commits , not compiling under linux 4.19.17

```
/tmp/build/rtl88XXau/rtl8812au/os_dep/linux/wifi_regd.c: En la función ‘_rtw_reg_apply_flags’:
/tmp/build/rtl88XXau/rtl8812au/os_dep/linux/wifi_regd.c:349:27: error: expected ‘;’ before ‘)’ token
       IEEE80211_CHAN_NO_IR);
                           ^
/tmp/build/rtl88XXau/rtl8812au/os_dep/linux/wifi_regd.c:349:27: error: expected statement before ‘)’ token
scripts/Makefile.build:303: fallo en las instrucciones para el objetivo '/tmp/build/rtl88XXau/rtl8812au/os_dep/linux/wifi_regd.o'
make[2]: *** [/tmp/build/rtl88XXau/rtl8812au/os_dep/linux/wifi_regd.o] Error 1
make[2]: *** Se espera a que terminen otras tareas....
Makefile:1520: fallo en las instrucciones para el objetivo '_module_/tmp/build/rtl88XXau/rtl8812au'
make[1]: *** [_module_/tmp/build/rtl88XXau/rtl8812au] Error 2
make[1]: se sale del directorio '/usr/src/linux-4.19.17'
Makefile:1550: fallo en las instrucciones para el objetivo 'modules'
make: *** [modules] Error 2
```

kimocoder, 2019-01-26:
Fixed. "git pull" and build. Thanks

closed 2019-01-26 by kimocoder

wifislax-ng, 2019-01-26:
It works now , thanks !!!