aircrack-ng/rtl8812au #264: [WORK-AROUND]  Dir name incorrect for DKMS install script after unzipping files

ckrzen, 2019-01-26:
Dir name is wrong after unzipping downloaded files:

~/Downloads/rtl8812au-5.2.20$ sudo ./dkms-install.sh 

About to run dkms install steps...
cp: cannot stat '../rtl8812au': No such file or directory
Error! Could not find module source directory.
Directory: /usr/src/rtl8812au-5.2.20.2 does not exist.
Error! Could not find module source directory.
Directory: /usr/src/rtl8812au-5.2.20.2 does not exist.
Error! Could not find module source directory.
Directory: /usr/src/rtl8812au-5.2.20.2 does not exist.
Finished running dkms install steps.

~/Downloads/rtl8812au-5.2.20$ cd ..

~/Downloads$ mv rtl8812au-5.2.20 rtl8812au

~/Downloads$ cd rtl8812au/

~/Downloads/rtl8812au$ sudo ./dkms-install.sh 

About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.2.20.2/source ->
                 /usr/src/rtl8812au-5.2.20.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.15.0-43-generic KSRC=/lib/modules/4.15.0-43-generic/build............
cleaning build area...

DKMS: build completed.

88XXau:
Running module version sanity check.
Error! Module version v5.2.20.2_28373.20180619 for 88XXau.ko
is not newer than what is already found in kernel 4.15.0-43-generic (v5.3.4_28226.20180614_COEX20171103-6c6d).
You may override by specifying --force.

depmod...

DKMS: install completed.
Finished running dkms install steps.

kimocoder, 2019-02-15:
Thanks. I'll add a fix shortly!

closed 2019-02-15 by kimocoder

kimocoder, 2019-02-15:
Fixed and updated in both v5.2.20 and v5.3.4 :1st_place_medal:

commit a5c39f36 (2019-02-15): Fixed DKMS path - fixes #264