aircrack-ng/rtl8812au #265: wpa_supplicant: Association request to the driver failed

ie04, 2019-01-27:
I have an extra wifi dongle hanging around, specifically the ObiWifi5g, which I used for a home phone setup but no longer need. I thought maybe I could repurpose it for a computer without a wifi adapter, which brought me to this driver. Compiling was relatively fine, and NetworkManager picked it up right away (on my Arch Linux machine). However, once I attempt to connect to a network connection, it tries to connect for a while before giving up. Investigating further, I used journalctl -f to see exactly what happens, which gave me this output:

 

> Jan 27 17:02:25 iArch wpa_supplicant[29449]: wlp0s20f0u1: Trying to associate with 8c:3b:ad:fa:a3:7a (SSID='wyfy' freq=5765 MHz)
> Jan 27 17:02:25 iArch wpa_supplicant[29449]: wlp0s20f0u1: Association request to the driver failed
> Jan 27 17:02:25 iArch wpa_supplicant[29449]: wlp0s20f0u1: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="wyfy" auth_failures=1 duration=10 reason=CONN_FAILED
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3219] device (wlp0s20f0u1): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3225] device (wlp0s20f0u1): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3228] device (wlp0s20f0u1): Activation: (wifi) access point 'wyfy 2' has security, but secrets are required.
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3228] device (wlp0s20f0u1): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3231] sup-iface[0x5640226b2590,wlp0s20f0u1]: wps: type pbc start...
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3673] device (wlp0s20f0u1): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3678] device (wlp0s20f0u1): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3682] device (wlp0s20f0u1): Activation: (wifi) connection 'wyfy 2' has security, and secrets exist.  No new secrets needed.
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3682] Config: added 'ssid' value 'wyfy'
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3682] Config: added 'scan_ssid' value '1'
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3682] Config: added 'bgscan' value 'simple:30:-80:86400'
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3682] Config: added 'key_mgmt' value 'WPA-PSK'
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3683] Config: added 'auth_alg' value 'OPEN'
> Jan 27 17:02:25 iArch NetworkManager[29425]: <info>  [1548626545.3683] Config: added 'psk' value '<hidden>'
> Jan 27 17:02:35 iArch NetworkManager[29425]: <info>  [1548626555.3884] device (wlp0s20f0u1): supplicant interface state: disconnected -> scanning
> Jan 27 17:02:35 iArch wpa_supplicant[29449]: wlp0s20f0u1: CTRL-EVENT-SSID-REENABLED id=0 ssid="wyfy"
> Jan 27 17:02:35 iArch wpa_supplicant[29449]: wlp0s20f0u1: Trying to associate with 8c:3b:ad:fa:a3:7a (SSID='wyfy' freq=5765 MHz)
> Jan 27 17:02:35 iArch wpa_supplicant[29449]: wlp0s20f0u1: Association request to the driver failed
> Jan 27 17:02:35 iArch wpa_supplicant[29449]: wlp0s20f0u1: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="wyfy" auth_failures=2 duration=20 reason=CONN_FAILED
> Jan 27 17:02:35 iArch NetworkManager[29425]: <info>  [1548626555.5690] device (wlp0s20f0u1): supplicant interface state: scanning -> disconnected

The crux of the issue appears to be wpa_supplicant, which fails to send an association request to the driver. I can replicate this same problem on two of my other machines using only wpa_supplicant, which still gives me the same error. Strangely enough, if I remove the dongle and put it back in, it connects automatically and functions perfectly. However, if I change the connection, the problem persists. I am unaware if this is a problem with the driver or on my end. Any idea?

arjunkc, 2019-01-30:
I have noticed a similar issue on my end with a Comfast card that has the 8812au chipset.

kimocoder, 2019-02-15:
@ie04 @arjunkc I've uploaded a bunch of documentation for hostapd and wpa_supplicant, it's a little bit tricky with these drivers unfortunately. using old nl80211/cfg80211 instead of the newer mac80211 and I/O handlers.

What branch where you using? I got reports that the v5.2.20 should work now, with right configuration of course (I added config files for examples too), but we're aiming for the v5.3.4 in a month or so.

Solution/tips/confirmations are welcome :+1:

arjunkc, 2019-02-15:
Unfortunately, I gave sent back my rtl8812au and cannot test for you, and settled on an older chipset that had more stable drivers.

ie04, 2019-02-17:
@kimocoder I'm using branch v5.2.20. I recompiled the driver and tried again, but I still get the same result. Is there any specific way I should configure wpa_supplicant?

kimocoder, 2019-02-17:
I don't use either hostapd or wpa_supplicant som I'm really not familier to them, but I added some documentation and configuration files to the "documentation" folder. For my understreking, these drivers is tricky to use either hostapd/wpa_sup.

You should take a look around, read read read :)

kimocoder, 2019-04-06:
Change to v5.3.4. All should do that now as the main issues are fixed and it's working better than the v5.2.20. If problems persists, open a new issue report. Thanks! :+1:

closed 2019-04-06 by kimocoder