aircrack-ng/rtl8812au #269: ArchLinux package available

noraj, 2019-02-06:
Hi,

I packaged the driver in v5.2.20 for ArchLinux: https://aur.archlinux.org/packages/rtl8812au-aircrack-ng-dkms-git/ and successfully built and use it for kernel 4.20.3-1-MANJARO.

I saw you added a kali supported badge, maybe you want one for archlinux/blackarch?

JpegXguy, 2019-02-10:
There's this now, I think it replaces the previous package nicely. IT also reflects that the drive is unified https://aur.archlinux.org/packages/rtl88xxau-aircrack-dkms-git/

noraj, 2019-02-10:
@JpegXguy Thanks. I didn't found it because the description and tags don't contains `rtl8812au`.

[image]

JpegXguy, 2019-02-10:
I see. We'd thought of 8812au being there in the search but not with the rtl, I'll make a comment

noraj, 2019-02-10:
@JpegXguy instead of a package deletion for my package why not a merge (see https://wiki.archlinux.org/index.php/Arch_User_Repository#Other_requests) ?

JpegXguy, 2019-02-10:
It didn't have any comments or votes and I didn't think of it. Merge is better in general, so you're right

kimocoder, 2019-02-15:
Awesome. Thanks for the heads up :v:

closed 2019-02-15 by kimocoder