aircrack-ng/rtl8812au #272: Problem with EDUP (rtl8814au) and hostapd

oangelo, 2019-02-09:
I am trying to use [EDUP](https://wikidevi.com/wiki/EDUP_EP-AC1621) USB3 card with hostapd. It works for a while, but a few minutes after handshake completed, it stops working.

This is the end part of hostapd debug msgs:

> nl80211: KEY_DATA - hexdump(len=16): [REMOVED]
   addr=18:56:80:6a:4c:08
wlan1: AP-STA-CONNECTED 18:56:80:6a:4c:08
nl80211: Set STA flags - ifname=wlan1 addr=18:56:80:6a:4c:08 total_flags=0x61 flags_or=0x1 flags_and=0xffffffff authorized=1
wlan1: STA 18:56:80:6a:4c:08 IEEE 802.1X: authorizing port
wlan1: STA 18:56:80:6a:4c:08 RADIUS: starting accounting session FEC79E5F52EEA23B
wlan1: STA 18:56:80:6a:4c:08 WPA: pairwise key handshake completed (RSN)
wlan1: ap_handle_timer: 18:56:80:6a:4c:08 flags=0x23 timeout_next=0
wlan1: Station 18:56:80:6a:4c:08 has been active 0s ago
ap_handle_timer: register ap_handle_timer timeout for 18:56:80:6a:4c:08 (303 seconds)

I am using Armbian on a [Lamobo-r1](http://linux-sunxi.org/Lamobo_R1):
> Linux Oraculo 4.14.84-sunxi #3 SMP Sat Dec 1 07:18:41 CET 2018 armv7l GNU/Linux

Is this a driver issue or a hostapd issue?

kimocoder, 2019-02-09:
This is a driver issue. Hostapd and wpa_supplicant got issues on branches below v5.3.4

oangelo, 2019-02-09:
@kimocoder, do you know if it is possible to use this hardware with hostapd with some workaround? Or, if there is any know driver that actually supports hostapd?

kimocoder, 2019-02-09:
You should take a look at [this](https://github.com/pritambaral/hostapd-rtl871xdrv), also take a look [here](https://github.com/igorpecovnik/hostapd)

You should also read https://github.com/kimocoder/rtl8812au/blob/v5.2.20/documents/Wireless_tools_porting_guide.pdf)

Also, the "documents" folder got some interesting information regarding the matter.
wpa_cli doc [here](https://github.com/kimocoder/rtl8812au/blob/v5.2.20/documents/wpa_cli_with_wpa_supplicant.pdf)

oangelo, 2019-02-10:
So, in practice is very hard to set hostapd for this hardware?
I compiled the hostapd from @igorpecovnik with rtl871xdrv and I could not make it work with this module.
Since I am not very knowledgeable in this matter, I think I have no other option but to give up.

Thanks for your time.

igorpecovnik, 2019-02-10:
> with rtl871xdrv

This is needed for some very old drivers. At least compile without this patch.

ggerla, 2019-02-10:
I didn't understand....
is the path needed also for rtl8814au?

igorpecovnik, 2019-02-10:
No, IIRC you don't need this particular patch for any of those.

oangelo, 2019-02-10:
I tried two drivers with the hostapd, both with rtl871xdrv and without, in Armbian and in Raspbian. There are so many repositories with drivers for rtl8814au that is impossible to test them all, and they are very poorly documented, I could not find any reference about the status of hostapd.

With rtl871xdrv I could not even start hostapd, without it, everything works for a few minutes before hostapd looses contact with the kernel module.

@igorpecovnik, are you aware about any situation where this card works with hostapd?

igorpecovnik, 2019-02-10:
I used to use 8814AU as my main AP for weeks but not with recent driver. I just started testing (armbian prepached hostapd and latest drivers compiled two days ago) and will try to reproduce ... I hope you power this device externally? Lamobo R1 is one of the most failed boards out there.

ggerla, 2019-02-10:
> I used to use 8814AU as my main AP for weeks but not with recent driver. I just started testing (armbian prepached hostapd and latest drivers compiled two days ago) and will try to reproduce ... I hope you power this device externally? Lamobo R1 is one of the most failed boards out there.

really?
I used AWUS1900 powered externally with Ubuntu 18.04. But the AP is up for 1 or 2 days and then I have to reboot the PC.
I used zebulon driver, this driver in version 5.2.20 and 5.3.4. Always the same result.
May depends on hostapd configuration?

igorpecovnik, 2019-02-10:
> really?

My summerhouse setup is DIY router based on https://www.armbian.com/orange-pi-plus-2e + powered USB hub. I also use Alfa AWUS1900 to have better range - I had zero issues, running Armbian with kernel 4.14.y which was rolling kernel in past summer. I use internal wireless for connecting to public wifi network and USB LTE modem as a backup. My internal network has up to 4 clients, all AC capable. Setup is stock Armbian, which uses patches hostapd, but mainly feature wise. In that time it was hostapd2.5, [configuration and patches](https://github.com/armbian/build/tree/master/packages/extras-buildpkgs/hostapd/debian/patches). Recently it was bumped to hostapd2.7

I am not aware how all this work on stock x86 Ubuntu+hostapd since I usually don't use them for this purpose.

oangelo, 2019-02-10:
I know that lamobo-r1 has a very bad reputation, but I am using it for more than a year as my only router and it works fine. I did change the onboard wifi to a RT5572 wireless adapter and it works great with hostapd. I am using external power for the rtl8814au, other wise the lamobo-r1 shuts down. 

Is the hostapd present in Armbian by "apt install hostapd" already patched or do I have to compile it? How do I do it? Can you share your hostapd.conf?

What bugs me is that it works for a short time, and if I close hostapd and start again, it works for a short time again. So frustrating...

ggerla, 2019-02-10:
> My summerhouse setup is DIY router based on https://www.armbian.com/orange-pi-plus-2e + 
> powered USB hub. I also use Alfa AWUS1900 to have better range - I had zero issues, running Armbian 
> with kernel 4.14.y which was rolling kernel in past summer. 

But Armbian has USB 2, not 3. This is first main difference...
Then on the usb hub I has also a AWUS036AC device.
So I have 2 instances of hostapd: one with AWUS036AC for 2.4GHz AP and the other with AWUS1900 for 5GHz AP.
I remember that I did some test with only one device connected but this didn't solve the problem.

So my question is: how are prformances with USB 2?
I have a lot of raspberry and I could try them.

igorpecovnik, 2019-02-10:
> Is the hostapd present in Armbian by "apt install hostapd" already patched or do I have to compile it?

That one is already patched. No need to do anything - use automated setup https://www.youtube.com/watch?v=__BJtInOQFY
> Can you share your hostapd.conf?

Generated automagically:
https://pastebin.com/srVk0KSL

> What bugs me is that it works for a short time, and if I close hostapd and start again, it works for a short time again. So frustrating...

I have this in the logs, https://pastebin.com/WKdPgec1 but it doesn't stop working.

> But Armbian has USB 2, not 3. This is first main difference...

USB3 is here due powering requirements, not due to bandwidth. OrangePi (Allwinner H2+/H3/H5) has 3 x dedicated USB2, 

> how are prformances with USB 2?

Client is Intel 7260 AC card running Debian Stretch and iperf3 shows 200-250Mbits/sec at proximity of few meters. (I am using 8814AU with internal antenna at this time)

> I have a lot of raspberry and I could try them.

Not sure about that. When it comes to I/O performances, Rpi is worse possible choice.

ggerla, 2019-02-10:
> USB3 is here mainly due powering requirements.

AWUS1900 has 1300Mbps at 5GHz... you cannot reach this speed with USB2 (max 480Mbs)
This is because I'm using an x86 platform.

I will check your hostapd conf file.

Thanks

kimocoder, 2019-02-10:
Remember.. most functions of this driver is unfortunately hardcoded..

How to cnfigure STA+SoftAP in 8812AU to work in 11ac mode
8812AU can support 11ac mode, which can offer a relatively high speed. if you want 
to enable 8812AU to work in 11ac mode. you need to modify the steps in SoftAP 
configuration above:
**modprobe -r 88XXau && modprobe 88XXau rtw_vht_enable=2**

or simply create a modprobe.d file for it om bootup
What’s more,the configuration file for 11ac mode is also rtl_hostapd_5.conf.

igorpecovnik, 2019-02-10:
> AWUS1900 has 1300Mbps at 5GHz... you cannot reach this speed with USB2

For my home use USB2 is enough. This above is fastest client I have, others are Phones and performing worse. When I will need more: Odroid XU4, Rock64, NanopiM4, ... boards with USB3 ports.

Show your iperf3 connection from your client to see if you can get better numbers.

> This is because I'm using an x86 platform.

That should have no impact on the problem.

ggerla, 2019-02-10:
> Remember.. most functions of this driver is unfortunately hardcoded..
> 
> How to cnfigure STA+SoftAP in 8812AU to work in 11ac mode
> 8812AU can support 11ac mode, which can offer a relatively high speed. if you want
> to enable 8812AU to work in 11ac mode. you need to modify the steps in SoftAP
> configuration above:
> **modprobe -r 88XXau && modprobe 88XXau rtw_vht_enable=2**
> 
> or simply create a modprobe.d file for it on bootup
> What’s more,the configuration file for 11ac mode is also rtl_hostapd_5.conf.
> 
> You'll find lots of info and configs in the "documentation" folder

I never used modprobe command... Usually I use dkms.
About the hostapd conf file I see some differences compared to mine.

You have driver=rtl871xdrv instead driver=nl80211 and I have vht configuration.

I will try to use rtl871xdrv in the future and I will inform you about results.

Thanks

kimocoder, 2019-02-10:
would be appreciated, as I personally don't have any use for hostapd, but lots of others do anyhow so it would have been nice with a spor on howto in the README.md.

Thanks

oangelo, 2019-02-26:
Just an update. I've tested my device on Archlinux on my pc (x86) and hostapd works fine for as long as I tested, a few hours.

I also bought another wifi dongle to test (Tp-link Archer T4U V3), it uses a different module (rtl8822bu) but it also works on my PC on Archlinux as ap. But I have the same problem on Armbian, the connection stops working after a few minutes.

Since they all work, the problem is not my hardware or the modules "per se". It might be a problem with those modules on ARM or with Armbian. But I have already hostapd with RT5572 Wireless Adapter working fine on my Armbian, but nothing else seems to work there...

Anyone have a hint of what might be? I am out of ideas...

igorpecovnik, 2019-02-27:
> https://forum.odroid.com/viewtopic.php?f=136&t=33634

Narrowing down a problem? We have updated hostapd, updated driver and updated kernel.

oangelo, 2019-02-27:
@igorpecovnik , I can't update my kernel on my Armbian because of the b53 switch driver. The switch is working fine with 4.14.84-sunxi kernel, I tried other kernels but the switch does not work. I do not have enough knowledge to  track this problem down... Do you know where I can find help? It might be a problem with a firmware or something else, but I can't figure it out. Newer kernel might solve the wifi problem, but I can't test it... :-(

oangelo, 2019-02-27:
About kernels. On my desktop, where everything seems to work (Tp-link Archer T4U V3 and EDUP ), I am using Arch Linux (x86) with 4.20.12-arch1-1-ARCH kernel. On Armbian both devices works for just a few minutes, the kernel version is  4.14.84-sunxi. On Raspbian, using kernels 4.19.25-v7+ and 4.14.79-v7+, I have the same behavior of Armbian, the devices works for a few minutes.

igorpecovnik, 2019-02-27:
> On Raspbian, using kernels 4.19.25-v7+ and 4.14.79-v7+, I have the same behavior of Armbian, the devices works for a few minutes.

This tells that there could be a regression in driver. I assume Raspbian uses stock Debian hostapd. Anoder idea is to perhaps blacklist driver that is a part of the kernel and trying this one https://github.com/aircrack-ng/rtl8812au/tree/v5.1.5 or some other branch.

oangelo, 2019-02-28:
@igorpecovnik, I got my Orange Pi PC+ from my 3D printer to test. Indeed, after updating Armbian, using 4.19.20-sunxi kernel,  the device (EDUP) works out of the box with [create_ap.](https://github.com/oblique/create_ap) Armbian is great! But is very strange that the device is not working on Raspbian with kernel 4.19.25-v7+ and neither on my lamobo-r1 with Armbian, where everything is up to date, except the kernel. It seems that an updated kernel with an patched hostapd is needed... Still, I have no idea of what is the problem.

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)