aircrack-ng/rtl8812au #279: AWUS036ACH dkms install on armbian bionic does not find signals in monitor mode

rhagu, 2019-02-12:
Hi,
I am running armbian bionic on a Nanopi Neo2 https://www.armbian.com/nanopi-neo-2/ and want to use my AWUS036ACH in monitor mode. The in kernel driver 88XXau was blacklisted in /etc/modprobe.d/blacklist.conf and ./dkms-install was performed without an error:

`sudo ./dkms-install.sh 
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.2.20.2/source ->
                 /usr/src/rtl8812au-5.2.20.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make' -j4 KVER=4.19.17-sunxi64 KSRC=/lib/modules/4.19.17-sunxi64/build........................................................................................
cleaning build area...

DKMS: build completed.

88XXau:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.17-sunxi64/updates/dkms/

depmod....

DKMS: install completed.
Finished running dkms install steps.
`
sudo dkms status shows that the module is up and running:
`rtl8812au, 5.2.20.2, 4.19.17-sunxi64, aarch64: installed`

and lsmod shows it as well:
`Module                  Size  Used by
zstd                   16384  4
8812au               1568768  0
zram                   24576  5
cfg80211              290816  1 8812au
snd_soc_simple_card    16384  0
sun8i_codec_analog     28672  0
snd_soc_simple_card_utils    16384  1 snd_soc_simple_card
sun4i_i2s              24576  0
sun8i_adda_pr_regmap    16384  1 sun8i_codec_analog
rfkill                 24576  2 cfg80211
snd_soc_core          126976  4 sun4i_i2s,sun8i_codec_analog,snd_soc_simple_card_utils,snd_soc_simple_card
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm                90112  3 sun4i_i2s,snd_soc_core,snd_pcm_dmaengine
snd_timer              32768  1 snd_pcm
snd                    57344  3 snd_timer,snd_soc_core,snd_pcm
sun4i_gpadc_iio        24576  0
soundcore              16384  1 snd
industrialio           53248  1 sun4i_gpadc_iio
cpufreq_dt             16384  0
thermal_sys            65536  2 cpufreq_dt,sun4i_gpadc_iio
sch_fq_codel           20480  6
usb_f_acm              16384  1
u_serial               24576  3 usb_f_acm
g_serial               16384  0
libcomposite           45056  2 g_serial,usb_f_acm
lima                   49152  0
gpu_sched              20480  1 lima
ttm                    86016  1 lima
realtek                20480  1
`
iwconfig recognizes the monitor mode:
`wlx00c0caa605e6  IEEE 802.11  Mode:Monitor  Frequency:2.442 GHz  Tx-Power=18 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off`

but kismet or airodump-ng just wont see any packets. I checked the Makefile, it is set to i368, should I change that? If so, shouldnt the compilation throw errors?
`###################### Platform Related #######################
CONFIG_PLATFORM_I386_PC = y
CONFIG_PLATFORM_ARM_RPI = n
CONFIG_PLATFORM_ARM64_RPI = n
CONFIG_PLATFORM_ANDROID_X86 = n
CONFIG_PLATFORM_ANDROID_INTEL_X86 = n
CONFIG_PLATFORM_JB_X86 = n
CONFIG_PLATFORM_ARM_S3C2K4 = n
CONFIG_PLATFORM_ARM_PXA2XX = n
CONFIG_PLATFORM_ARM_S3C6K4 = n
CONFIG_PLATFORM_MIPS_RMI = n
CONFIG_PLATFORM_RTD2880B = n
CONFIG_PLATFORM_MIPS_AR9132 = n
CONFIG_PLATFORM_RTK_DMP = n
CONFIG_PLATFORM_MIPS_PLM = n
CONFIG_PLATFORM_MSTAR389 = n
CONFIG_PLATFORM_MT53XX = n
CONFIG_PLATFORM_ARM_MX51_241H = n
CONFIG_PLATFORM_FS_MX61 = n
CONFIG_PLATFORM_ACTIONS_ATJ227X = n
CONFIG_PLATFORM_TEGRA3_CARDHU = n
CONFIG_PLATFORM_TEGRA4_DALMORE = n
CONFIG_PLATFORM_ARM_TCC8900 = n
CONFIG_PLATFORM_ARM_TCC8920 = n
CONFIG_PLATFORM_ARM_TCC8920_JB42 = n
CONFIG_PLATFORM_ARM_TCC8930_JB42 = n
CONFIG_PLATFORM_ARM_RK2818 = n
CONFIG_PLATFORM_ARM_RK3066 = n
CONFIG_PLATFORM_ARM_RK3188 = n
CONFIG_PLATFORM_ARM_URBETTER = n
CONFIG_PLATFORM_ARM_TI_PANDA = n
CONFIG_PLATFORM_MIPS_JZ4760 = n
CONFIG_PLATFORM_DMP_PHILIPS = n
CONFIG_PLATFORM_MSTAR_TITANIA12 = n
CONFIG_PLATFORM_MSTAR = n
CONFIG_PLATFORM_SZEBOOK = n
CONFIG_PLATFORM_ARM_SUNxI = n
CONFIG_PLATFORM_ARM_SUN6I = n
CONFIG_PLATFORM_ARM_SUN7I = n
CONFIG_PLATFORM_ARM_SUN8I_W3P1 = n
CONFIG_PLATFORM_ARM_SUN8I_W5P1 = n
CONFIG_PLATFORM_ACTIONS_ATM702X = n
CONFIG_PLATFORM_ACTIONS_ATV5201 = n
CONFIG_PLATFORM_ACTIONS_ATM705X = n
CONFIG_PLATFORM_ARM_SUN50IW1P1 = n
CONFIG_PLATFORM_ARM_RTD299X = n
CONFIG_PLATFORM_ARM_SPREADTRUM_6820 = n
CONFIG_PLATFORM_ARM_SPREADTRUM_8810 = n
CONFIG_PLATFORM_ARM_WMT = n
CONFIG_PLATFORM_TI_DM365 = n
CONFIG_PLATFORM_MOZART = n
CONFIG_PLATFORM_RTK119X = n
CONFIG_PLATFORM_RTK129X = n
CONFIG_PLATFORM_NOVATEK_NT72668 = n
CONFIG_PLATFORM_HISILICON = n
CONFIG_PLATFORM_NV_TK1 = n
CONFIG_PLATFORM_RTL8197D = n
CONFIG_PLATFORM_ARM_ODROIDC2 = n
CONFIG_PLATFORM_PPC = n
`

Chadster766, 2019-02-17:
Try upgrade to v5.3.4 which iwconfig will not work with instead use "ip link" command to set promiscuous mode and monitor mode.

rhagu, 2019-02-23:
I switched to armbian stretch and it works out of the box now. I am sorry I cannot help other in this situation but this fixed the problem for me.

closed 2019-02-23 by rhagu