aircrack-ng/rtl8812au #284: Can't compile under kernel ver. 4.19

cyberhoffman, 2019-02-20:


cyberhoffman, 2019-02-20:
Hi! First of all I would like to thank you for such great work. The driver has been worked perfectly till today. Today I've upgraded my linux kernel version from 4.14.83 to 4.19.23 and now I've got a problem with compiling "rtl8812au_aircrack-ng-5.2.20". I guess that the issue is similar to https://github.com/smlinux/rtl8723de/issues/27 . I leave here my full build log: https://paste.pound-python.org/show/6JYNyJdDlAR9zCZkq1r5/ and hope it will help to create a patch. Thank you!

kimocoder, 2019-02-22:
These drivers are working with kernel up to v5.0. Also v4.19 and v4.20.
Post logs please, cause the logs you provided, leads to a 404 address

cyberhoffman, 2019-02-25:
> These drivers are working with kernel up to v5.0. Also v4.19 and v4.20.
> Post logs please, cause the logs you provided, leads to a 404 address

Thanks for your reply. Indeed, that link is broken.  I've just created new one: http://dpaste.com/11CR85R

closed 2019-04-06 by kimocoder