aircrack-ng/rtl8812au #287: Driver installation for AWUS1900 with Kali rolling on RPi 3+ fails consistently.

peerDE, 2019-02-28:
Installing the driver for USB wifi network card fails. To reproduce:

# apt-get install realtek-rtl88xxau-dkms

The bug was already filed on kali's bugtracker, where I was asked to file it here.
For details (installation protocolls and error messages) see:

https://bugs.kali.org/view.php?id=5282

Re4son, 2019-03-01:
The dkms package has some issues with the Raspberry Pi. It shouldn't matter though as the kali image comes with that driver pre-installed so it should be a plug-and-play affair.
Just plug it in and give it a go.

peerDE, 2019-03-01:
In other words, I can use the AWUS1900 with a freshly installed image, but the dkms package on Kali's RPi image can't compile the driver, so once a newer kernel get's installed via kali rolling, the driver gets out of sync with the kernel and might stop to function correctly, because the dkms recompilation is supposed to work but doesn't work, and then it's kind of impossible to get it right without getting kernel sources and so on on the RPi to fix the broken-on-RPi dkms installer's work. Is this correct?

Re4son, 2019-03-01:
Nope :-)
The driver is part of the kernel tree, thus if you update the kernel you'll get the matching driver too.

peerDE, 2019-03-01:
In these instructions, it's the step:
$ sudo apt-get install linux-headers-`uname -r`
which doesn't work giving the error: "Unable to locate package linux-headers-4.14.71-v8", so there is an instruction bug in the doc if you try this on the RPi.

$ sudo apt-cache search linux-headers-* shows only several versions "4.19.0-kali3-"something.
So the versions did indeed get out of sync. I installed the 2018.4 kali just some days before 2019.1 came out, and rolled onto the 2019.1.

So now I tried my luck with
$ sudo apt-get install linux-headers-4.19.0-kali3-arm64
which at the end tried to compile the hanging (non-installed due to these errors) driver, but still it doesn't compile, although the error message is different. It doesn't complain about missing headers any more.
The log now says:
```
DKMS make.log for realtek-rtl88xxau-5.2.20.2~20181106 for kernel 4.19.0-kali3-arm64 (aarch64)
Fri Mar  1 08:30:23 UTC 2019
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.0-kali3-arm64/build M=/var/lib/dkms/realtek-rtl88xxau/5.2.20.2~20181106/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-kali3-arm64'
  CC [M]  /var/lib/dkms/realtek-rtl88xxau/5.2.20.2~20181106/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/realtek-rtl88xxau/5.2.20.2~20181106/build/core/rtw_security.o
gcc-8: error: unrecognized command line option ‘-mhard-float’; did you mean ‘--warn-format’?
make[4]: *** [/usr/src/linux-headers-4.19.0-kali3-common/scripts/Makefile.build:309: /var/lib/dkms/realtek-rtl88xxau/5.2.20.2~20181106/build/core/rtw_cmd.o] Error 1
make[4]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/realtek-rtl88xxau/5.2.20.2~20181106/build/core/rtw_debug.o
gcc-8: error: unrecognized command line option ‘-mhard-float’; did you mean ‘--warn-format’?
make[4]: *** [/usr/src/linux-headers-4.19.0-kali3-common/scripts/Makefile.build:309: /var/lib/dkms/realtek-rtl88xxau/5.2.20.2~20181106/build/core/rtw_security.o] Error 1
gcc-8: error: unrecognized command line option ‘-mhard-float’; did you mean ‘--warn-format’?
make[4]: *** [/usr/src/linux-headers-4.19.0-kali3-common/scripts/Makefile.build:309: /var/lib/dkms/realtek-rtl88xxau/5.2.20.2~20181106/build/core/rtw_debug.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.19.0-kali3-common/Makefile:1535: _module_/var/lib/dkms/realtek-rtl88xxau/5.2.20.2~20181106/build] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-kali3-arm64'
make: *** [Makefile:1935: modules] Error 2

```
I wonder if the problem is due to the CROSS_COMPILE thing, being on the platform I'm compiling for.

peerDE, 2019-03-01:
For some reason, "airmon-ng" shows only the Broadcom 43430 Driver.
Maybe I killed the driver delivered with the new kernel by trying different sources and methods from the web.
OTOH, kimocoder filed this information https://bugs.kali.org/view.php?id=5283 leading to this change https://github.com/aircrack-ng/rtl8812au/blob/v5.2.20/realtek_88XXau.conf which changed a typo (881XXau => 88XXau), maybe this type (before the correction) prevented the inclusion of the driver in the original image? Or maybe it was purged out by apt?
I suppose my options now are to (1) try harder! to understand the mechanisms and get the compilation running, or (2) start anew with a fresh image.

peerDE, 2019-03-01:
There is a /lib/modules/4.19.0-kali3-arm64/updates/88XXau.ko, but
modprobe 88XXau
complains that it doesn't find it in "/lib/modules/4.14.71-v8. A
find /lib/modules/4.14.71-v8 -name "*88*"
confirms that such a driver is not there. Also, a
insmod /lib/modules/4.19.0-kali3-arm64/updates/88XXau.ko
complains about an invalid module format. Which is no wonder because
uname -r
still shows I'm running 4.14.71-v8.
So why did the search for the headers (see above) only offer headers for 4.19.0?
I'm getting somewhat confused, but I guess the desync seems to come from the rolling process creating somehow incompatible versions, for kernels, os-release, and modules.
cat /etc/os-release
shows VERSION="2019.1"  and VERSIONS_ID="2019.1"
cat /etc/apt/sources.list
shows two lines (deb and deb-src) for "kali-rolling".

Re4son, 2019-03-01:
Not sure. Let me dig into that.
The arm64 image is brand new and still in an experimental stage. Some drivers haven't been fully tested yet. That's the purpose of this image, to test it all out. I'd go with the armhf image for now if you can't get it to work.
I just did a manual installation on armhf via git clone, make & make install, and it worked like a charm

Re4son, 2019-03-01:
I just checked with a brand new image and it all seems to work:
root@kali:~# ls /lib/modules/4.14.93-Re4son-v8+/kernel/drivers/net/wireless/realtek/rtl8812au
88XXau.ko
root@kali:~# modprobe 88XXau
root@kali:~# lsmod
Module                  Size  Used by
88XXau               2498560  0

I think that the installation of those additional kernel headers and dkms dependencies threw your installation off balance.
Sorry for the trouble, we'll update the documentation.

peerDE, 2019-03-01:
Maybe this helps, speaking about the platform differences regarding the floating point error my compilation produced:
https://stackoverflow.com/questions/37790029/what-is-difference-between-arm64-and-armhf

Maybe I could try to tune the options, but as for me, it would then produce a 4.19 module which my 4.14 kernel won't load.

The missing module in 4.14.71-v8 might be due to the typo only, but I don't understand what the typo in the options of the dkms install script results in, so this is to a more competent person to evaluate.

Anyway, running a 4.14.71 kernel on a platform which otherwise thinks it's 4.19 also doesn't look exactly correct. So how come the kali rolling mechanism did not install the new kernel (or it's boot mechanism)?

Re4son, 2019-03-01:
I've just compiled the driver manually without any problems on a new arm64 image:

```
root@kali:~/rtl8812au# make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.14.93-Re4son-v8+/build M=/root/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.93-Re4son-v8+'
  CC [M]  /root/rtl8812au/core/rtw_cmd.o
<snip>
...
<snip>
Building modules, stage 2.
  MODPOST 1 modules
  CC      /root/rtl8812au/88XXau.mod.o
  LD [M]  /root/rtl8812au/88XXau.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.14.93-Re4son-v8+'
```

Having said that, the module is identical to the one shipped with the kernel.

I conclude that:
- The kali images support the adapter out of the box. It's plug and play and does not require any additional drivers
- The stock kali installation is set up correctly and exhaustive to allow compiling kernel drivers without any additional packages apart from "bc"
- We need to update the driver documentation to point out that the dkms package is not required for Kali-Pi's and not supported either.

Sorry for the trouble and thanks for helping us getting that sorted.

peerDE, 2019-03-01:
Ok, great. In the meantime I tried to install the new kernel image for arm64, but this results in some errors:

```
root@kali:~# apt-get install linux-image-arm64
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  apparmor irqbalance linux-image-4.19.0-kali3-arm64 runit-helper
Suggested packages:
  apparmor-profiles-extra apparmor-utils linux-doc-4.19 debian-kernel-handbook
The following NEW packages will be installed:
  apparmor irqbalance linux-image-4.19.0-kali3-arm64 linux-image-arm64 runit-helper
0 upgraded, 5 newly installed, 0 to remove and 2 not upgraded.
1 not fully installed or removed.
Need to get 40.1 MB of archives.
After this operation, 223 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://ftp.halifax.rwth-aachen.de/kali kali-rolling/main arm64 apparmor arm64 2.13.2-7 [501 kB]
Get:2 http://ftp.halifax.rwth-aachen.de/kali kali-rolling/main arm64 runit-helper all 2.8.6 [4900 B]
Get:3 http://ftp.halifax.rwth-aachen.de/kali kali-rolling/main arm64 irqbalance arm64 1.5.0-3 [57.3 kB]
Get:4 http://ftp.halifax.rwth-aachen.de/kali kali-rolling/main arm64 linux-image-4.19.0-kali3-arm64 arm64 4.19.20-1kali1 [39.5 MB]
Get:4 http://ftp.halifax.rwth-aachen.de/kali kali-rolling/main arm64 linux-image-4.19.0-kali3-arm64 arm64 4.19.20-1kali1 [39.5 MB]
Get:4 http://ftp.halifax.rwth-aachen.de/kali kali-rolling/main arm64 linux-image-4.19.0-kali3-arm64 arm64 4.19.20-1kali1 [39.5 MB]
Get:4 http://ftp.halifax.rwth-aachen.de/kali kali-rolling/main arm64 linux-image-4.19.0-kali3-arm64 arm64 4.19.20-1kali1 [39.5 MB]
Get:5 http://ftp.halifax.rwth-aachen.de/kali kali-rolling/main arm64 linux-image-arm64 arm64 4.19+103+kali1 [8896 B]
Fetched 1074 kB in 3min 49s (4696 B/s)      
Preconfiguring packages ...
Selecting previously unselected package apparmor.
(Reading database ... 341175 files and directories currently installed.)
Preparing to unpack .../apparmor_2.13.2-7_arm64.deb ...
Unpacking apparmor (2.13.2-7) ...
Selecting previously unselected package runit-helper.
Preparing to unpack .../runit-helper_2.8.6_all.deb ...
Unpacking runit-helper (2.8.6) ...
Selecting previously unselected package irqbalance.
Preparing to unpack .../irqbalance_1.5.0-3_arm64.deb ...
Unpacking irqbalance (1.5.0-3) ...
Selecting previously unselected package linux-image-4.19.0-kali3-arm64.
Preparing to unpack .../linux-image-4.19.0-kali3-arm64_4.19.20-1kali1_arm64.deb ...
Unpacking linux-image-4.19.0-kali3-arm64 (4.19.20-1kali1) ...
Selecting previously unselected package linux-image-arm64.
Preparing to unpack .../linux-image-arm64_4.19+103+kali1_arm64.deb ...
Unpacking linux-image-arm64 (4.19+103+kali1) ...
Setting up runit-helper (2.8.6) ...
Setting up irqbalance (1.5.0-3) ...
update-rc.d: We have no instructions for the irqbalance init script.
update-rc.d: It looks like a non-network service, we enable it.
Setting up realtek-rtl88xxau-dkms (5.2.20.2~20181106-0kali1) ...
Removing old realtek-rtl88xxau-5.2.20.2~20181106 DKMS files...

------------------------------
Deleting module version: 5.2.20.2~20181106
completely from the DKMS tree.
------------------------------
Done.
Loading new realtek-rtl88xxau-5.2.20.2~20181106 DKMS files...
It is likely that 4.14.71-v8 belongs to a chroot's host
Building for 4.14.71-v8 and 4.19.0-kali3-arm64
Building initial module for 4.14.71-v8
Error! Bad return status for module build on kernel: 4.14.71-v8 (aarch64)
Consult /var/lib/dkms/realtek-rtl88xxau/5.2.20.2~20181106/build/make.log for more information.
dpkg: error processing package realtek-rtl88xxau-dkms (--configure):
 installed realtek-rtl88xxau-dkms package post-installation script subprocess returned error exit status 10
Setting up apparmor (2.13.2-7) ...
update-rc.d: We have no instructions for the apparmor init script.
update-rc.d: It looks like a non-network service, we enable it.
Setting up linux-image-4.19.0-kali3-arm64 (4.19.20-1kali1) ...
I: /vmlinuz.old is now a symlink to boot/vmlinuz-4.19.0-kali3-arm64
I: /initrd.img.old is now a symlink to boot/initrd.img-4.19.0-kali3-arm64
[... 13 lines cut ...]
pigz: abort: write error on <stdout> (No space left on device)
E: mkinitramfs failure cpio 141 pigz 28
update-initramfs: failed for /boot/initrd.img-4.19.0-kali3-arm64 with 1.
run-parts: /etc/kernel/postinst.d/initramfs-tools exited with return code 1
dpkg: error processing package linux-image-4.19.0-kali3-arm64 (--configure):
 installed linux-image-4.19.0-kali3-arm64 package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of linux-image-arm64:
 linux-image-arm64 depends on linux-image-4.19.0-kali3-arm64; however:
  Package linux-image-4.19.0-kali3-arm64 is not configured yet.

dpkg: error processing package linux-image-arm64 (--configure):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for systemd (240-6) ...
Errors were encountered while processing:
 realtek-rtl88xxau-dkms
 linux-image-4.19.0-kali3-arm64
 linux-image-arm64
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@kali:~# 
```
I can still boot but get the old kernel. Maybe this log helps to solve some other problems with installing a newer image while on the arm64 platform on a RPi.
But I'm not sure if trying harder with this setup still makes sens now for you. I'm happy with your results (and reaction), so no problem on this side
My current setup is more for testing and I won't be hurt if I just discard it, and the adapter is for another hw anyway, so I'll set that up with a new (and bigger) SD card, with a newer image and kernel.
So if you want to close the bug that's fine with me, or I could test something else if you want to suggest something. Otherwise I'd start over with a newer image for my test platform as well.

Re4son, 2019-03-01:
One more thing: The raspberry pi is very unique and required us to create separate kernel and firmware packages so the standard arm64 ones won't do it. 
We'll close the bug. Thanks again for helping us to improve our documentation.

peerDE, 2019-03-01:
Ok, thank you as well!

closed 2019-03-01 by peerDE