aircrack-ng/rtl8812au #288: Makefile 'uname' overrides dkms variables on x386 build

minsonj2016, 2019-02-28:
Pardon me if I am incorrect but will not the entries at lines 682 and 683 override the values of 'KVER' and 'KSRC' being supplied by the 'dkms.conf' file ? This will cause the module to be built for the running kernel not the 1 just installed .

I am on Fedora29  X86 arch .

Otherwise thanks for keeping this driver alive as many of the other 'gits' seem to have fallen by the wayside .

minsonj2016, 2019-02-28:
how about

KVER  ?= $(shell uname -r)
KSRC  ?= /lib/modules/$(KVER)/build

kimocoder, 2019-04-07:
@minsonj2016 you could fork the driver and test/add it yourself, so credits go where they should go :+1:

kimocoder, 2019-11-14:
This issue should be resolved. We've moved over to v5.6.4.2 branch, all the older will be left behind. Thanks

closed 2019-11-14 by kimocoder