aircrack-ng/rtl8812au #290: AWUS036ACH Very poorly working on MacBook Pro

4val0v, 2019-03-05:
Hello, there is _AWUS036ACH_ but it works differently on different PCs.
**OS:** Kali Linux 2019.1
**Driver** _v5.2.20.2_ installed from repository: `apt install realtek-rtl88xxau-dkms`

**MacBook:** - _My phone is a lot more wi-fi points sees ;)_
[image]
[image]

**Samsung NP270:** - _More wi-fi access points_ **_(Sees +9 more networks)_**
[image]
[image]

How do I understand the problem because of the power supplied by the USB port ?

4val0v, 2019-03-06:
In the client mode does not work long _(after measuring the speed of the Internet, loses connection with the wifi point)_, see video.
The adapter only sees points near :(
https://youtu.be/cJScuHW-KSw

Log data from UM34C - [UM34C.xls.zip](https://github.com/aircrack-ng/rtl8812au/files/2936737/UM34C.xls.zip)

LinSSID sees more wi-fi points, but also far from ideal:
[image]

_Checked on MacBook Pro, if needed I can check on another device_

kimocoder, 2019-11-14:
This issue should be resolved on our newest branch v5.6.4.2, if not feel free to re-open this issue again. Thanks!

closed 2019-11-14 by kimocoder