aircrack-ng/rtl8812au #293: ALFA AWUS036AC: Association request to the driver failed (raspian stretch)

0unknwn, 2019-03-07:
I'm using a RasPi 3 B + (latest and up to date raspbian stretch) with `systemd-networkd` and` wpa_supplicant`. When I restart the wpa_supplicant@wlan1.service I get a ` Association request to the driver failed`. I don't have this issue with other wifi dongles. What could be the reason?

chryzsh, 2019-03-30:
I installed from branch v5.4.3 and that solved this issue. However, wpa doesn't work in this version (either).

0unknwn, 2019-04-04:
The error is gone with v5.4.3 for me as well. And everything works fine!

kimocoder, 2019-04-04:
We move over to the v5.3.4 in a while, some issues with reaver has been solved.
Please just be patient and I'll start on it.

kimocoder, 2019-04-06:
Now the v5.3.4 branch has got several fixes added, including some hostapd fixes from @j-forestial
Also, there was a auth_seq and other minors that has been fixed.

It should be working better now, as we move over to v5.3.4. If not, open a new report!
Thanks! :+1:

closed 2019-04-06 by kimocoder