aircrack-ng/rtl8812au #294: There is no proc/net/rtl8812au/$(your interface name)/led_enable

0unknwn, 2019-03-07:
Installed the latest driver on the latest and up to date raspbian stretch (RPI 3 B +) and missing the ability to control the LED. What could be the reason? How to fix it?

gordboy, 2019-04-01:
As far as I am aware, only the 5.2.20.2 branch has dynamic LED support, which was one of the few changes I made to the functionality of the original Realtek code.

And you need

```
CONFIG_LED_CONTROL = y
CONFIG_LED_ENABLE = y
```

in the Makefile.

I was vaguely thinking of submitting a patch/pull request to fix this for 5.3.4 but never really found the time. The changes needed are not trivial, and other stuff came first.

0unknwn, 2019-04-04:
Thanks for your answer! It's not that important, but was a nice feature.

closed 2019-04-07 by kimocoder