aircrack-ng/rtl8812au #295: how to clone 5.3.4 from 1 month ago?

logan001, 2019-03-08:
i had to reinstall linuxmint and reinstall the driver to create softap with hostapd. for some strange reason i could get only 5Mbits download on devices connected to my softap. and thats over a speedtest hosted on the same computer. with another driver i can above 100Mbits/s (~110) but that drivers has no dkms installer.
How do i clone this 5.3.4 version from a previous date to test the speed?.
I google it a bit and found something about git checkout but i have no clue how to set the git to clone and also the specific version (branch?)

ValZapod, 2019-03-09:
When you download git repo, you download all history with it.... You can indeed use 

> git checkout commit_hash 

to checkout to any version before... just look at commit hashes https://github.com/aircrack-ng/rtl8812au/commits/v5.3.4 for example 3c48dff for the first commit...

logan001, 2019-03-09:
i finally manage to do it. what i didn't know was when/where to use the git checkout.
everywhere i've look nobody said that i have to go in the downloaded folder. i have no clue how this git works but if i have to guess it seems that it downloads everything localy and can undo/redo everything inside that folder and a lot more.
well now with 2d0547c i can use softap with 100+ Mb/s again. thats the only one i'\ve tried other that the main download. i didn't bother to check every coming to find out when the problem occurred.

closed 2019-03-09 by aircrack-ng