aircrack-ng/rtl8812au #296: Difference between this repo and others?

tech4him1, 2019-03-13:
Is there any intentional use case or difference between this repo, gordboy/rtl8812au, etc? Thanks!

gordboy, 2019-04-01:
gordboy repo has far fewer changes from the Realtek original 5.2.20.2 code, and it does not support monitor mode or any other chipset than 8812au.

The basic idea was to not add any features, and just fix compilation every time the kernel API changes. I did not stick to that 100% but pretty close.

Also, gordboy repo is more 'stable' in the sense that there are no surprises every few days. If it was broken before, it stays broken :)

tech4him1, 2019-04-02:
@gordboy That makes sense, thanks for your input here. Was there a particular reason other chipsets were not supported (or just your time 😄) ?

huitc, 2019-04-04:
@gordboy 
```
...
  CC [M]  /root/Desktop/gordboy/hal/rtl8812a/usb/rtl8812au_xmit.o
/root/Desktop/gordboy/hal/rtl8812a/usb/rtl8812au_xmit.c: In function ‘upload_txpktbuf_8812au’:
/root/Desktop/gordboy/hal/rtl8812a/usb/rtl8812au_xmit.c:418:61: warning: bitwise comparison always evaluates to false [-Wtautological-compare]
   while ((rtw_read32(adapter, REG_PKTBUF_DBG_CTRL) & BIT23) == 1) {
                                                             ^~
  CC [M]  /root/Desktop/gordboy/hal/rtl8812a/usb/rtl8812au_recv.o
...
```
`Linux kali 4.19.0-kali4-amd64 #1 SMP Debian 4.19.28-2kali1 (2019-03-18) x86_64 GNU/Linux`
`win10 pro` `vbox 6.0.4 r128413`

tech4him1, 2019-04-04:
@huitc That doesn't seem to be related to this issue.

visbs, 2019-04-09:
@gordboy  can u see my issue #314

closed 2019-04-09 by tech4him1