aircrack-ng/rtl8812au #297: Does it support kernel 4.19?

barakat-abweh, 2019-03-14:


minsonj2016, 2019-03-14:
Works on Linux linuxdesktop 4.20.14-200.fc29.x86_64

barakat-abweh, 2019-03-14:
@minsonj2016 
I'm asking because the driver makes my laptop freezing each time I install it, I'm using Debian 10(Buster) with kernel 4.19.0-2-amd64, also I'm facing another problem when trying to run airmon-ng and aireplay-ng --test, the airmon-ng problem with awus1900 can be bypassed with ifconfig int# down&&iwconfig int# mode monitor, but the problem with aireplay-ng --test is it doesn't really work with this driver

kimocoder, 2019-04-06:
First off all.. this driver works up to kernel v5.0...

If you read the README.md properly, airmon-ng isn't really supported (a dirty fix has been added, but I would not use it.)

And you use iwconfig which is deprecated, READ the README.md, use "iw" instead.
And try avoid using airmon-ng, cause sometimes it makes the driver hang.

README.md

closed 2019-04-06 by kimocoder