aircrack-ng/rtl8812au #298: Support for radiotap headers/RSSI data can't be seen in pcap files

charliebabbitt1988, 2019-03-15:
I'm not sure why, but I am not capturing RSSI data in cap files with this driver.  That stated, I seem to have 2 different versions of this driver loaded.  I have the older 5.1.5 driver and the newer 5.2.20 driver.

Both drivers can be seen with "modinfo 8812au" and "modinfo 88XXau".  Should I uninstall one or both of these drivers and retry capturing a pcap file?

I am testing these cap files with this command:

tshark -nr capture.cap -T fields -e wlan.sa -e wlan.ssid -e wlan_radio.signal_dbm

The 3rd column, wlan_radio.signal_dbm, has no data, but previous tests had some data for that field.

Also, I can't seem to locate the RSSI data in Wireshark.  The GUI doesn't help.

Is there a way to purge both of these drivers and start fresh?

charliebabbitt1988, 2019-03-16:
Turns out that this is a limitation of airodump-ng.

https://github.com/aircrack-ng/aircrack-ng/issues/1883
https://github.com/aircrack-ng/aircrack-ng/issues/2012

I keep reading that mods to "osdep" would be necessary.
https://github.com/aircrack-ng/aircrack-ng/tree/master/src/aircrack-osdep

It seems that I will have to find a different capture method if I want 802.11 control frames in a pcap file.

aircrack-ng, 2019-03-16:
tshark and wireshark aren't part of aircrack-ng.

Data not present in the radiotap is a limitation of the driver/firmware/card

Aircrack-ng captures control frames just fine. If they're not present, it's a limitation of the driver/firmware/hardware. So, there is no modifications to aircrack-ng/osdep necessary to get them.

charliebabbitt1988, 2019-03-16:
After capturing some data to a cap/pcap file, when I try execute, "airodump-ng -r capture.cap", the data for the RSSI isn't there because it seemingly hasn't been recorded in the cap/pcap file.

The adapter I am using is the AWUS036ACH with the v5.1.5 driver that I found here.  I verified this with:
modinfo 8812au | grep ^'version'

The output was: v5.1.5_19247.20160830

The workaround that I have come up with is to use tshark (I know, not part of aircrack-ng, but I can  capture the RSSI data) like this:

ip link set $interface down
iw dev $interface set type monitor
ip link set $interface up
iw dev $interface set channel 11
tshark -i $interface -w tshark_test.pcap -a duration:30

The resulting capture file with tshark has populated values for wlan_radio.signal_dbm, but the capture file with airodump-ng doesn't.  I don't know why, but the only control frames that have an RSSI are the "Block Ack" control frames (wlan.fc.type_subtype==25).

Note that I am using the same driver for both tests, one test with airodump-ng and the other with tshark.

This is why I concluded that the driver is probably fine, but this behavior had to be a limitation of airodump-ng.  I could be wrong.

If airodump-ng supports the RSSI data being written to a capture file, then could this be a limitation of the driver?

I really don't know where the problem is now.

aircrack-ng, 2019-03-16:
I can't remember the details but I know there was a bug reading signal from PCAP. I don't remember exactly when it has been fixed, but current master has the fix for sure.

Two things:
1. Is the RSSI present when opening the capture file with Wireshark?
2. Is the RSSI present when processing the file with current airodump-ng (git master)?

charliebabbitt1988, 2019-03-17:
I am running Ubuntu 18.10, so I needed to install some Debian packages in order to get the latest aircrack-ng.

After getting an updated libhwloc5 Debian package install from here:
ftp.br.debian.org/debian/pool/main/h/hwloc/libhwloc5_1.11.12-3_amd64.deb

I installed it with "dpkg -i"

and I installed aircrack-ng from here:
ftp.br.debian.org/debian/pool/main/a/aircrack-ng/aircrack-ng_1.5.2-3_amd64.deb

The aircrack-ng debian package wouldn't install without the libhwloc5_1.11.12-3_amd64.deb Debian package.

I also installed the "ieee_data" package with "sudo apt-get install ieee_data"

The ieee_data package was one of the "Recommends" when I attempted to install aircrack-ng and it didn't work due to the absence of the libhwloc5_1.11.12-3_amd64.deb Debian package.

I verified the version of airodump-ng and it says, "Airodump-ng 1.5.2" at the top of the output when I execute "airodump-ng --help"

Having repeated the capture tests, one with airodump-ng and one with tshark, I get better results when testing with tshark.  I can now see Radiotap header data (RSSI) for more than just the "Block Ack" control frames.  I'm not sure what improved the result with tshark, but it might have been the ieee_data package.

The capture file that airodump-ng produces, however, has the same result.  There is no Radiotap header data in the capture file.

Also, I checked the capture file that airodump-ng produces with Wireshark and confirmed that it didn't have that data.  The "Radiotap Header v0, Length 32" section wasn't available for anything captured with airodump-ng.

Conclusion?  The driver seems okay, but perhaps I am missing another Debian package?  Maybe I need to update one of the Debian packages?

Or could the culprit be a limitation of airodump-ng?

aircrack-ng, 2019-03-17:
Just FYI, we have packages in packagecloud.io, so you don't have to install any weird stuff: https://packagecloud.io/aircrack-ng/release

Doing it the way you are currently doing will certainly break your system sooner or later.

What I was asking is when you capture with tshark, then feed it to wireshark/airodump-ng.

charliebabbitt1988, 2019-03-17:
I'll definitely check out those packages.  Thanks.

I can open the tshark produced capture file when using Wireshark.  Wireshark can open both the tshark produced capture file as well as the airodump-ng produced capture file.

When trying to open the same capture file with "airodump-ng -r capture.cap", I get an error message that says:
"capture.cap" isn't a pcap file (expected TCPDUMP_MAGIC).

aircrack-ng, 2019-03-18:
You probably saved it as pcapng. Save it as pcap and it will work.

So:
1. Does wireshark report the signal in the radiotap headers?
2. Does airodump-ng report the signal using that capture file?

charliebabbitt1988, 2019-03-19:
I tried airodump-ng again with "airodump-ng --band abg $interface --output-format pcap -w capfile.pcap".  (I get an automatic suffix of "-01.cap", normal behavior of airodump-ng, it seems.)

1. I tried Wireshark again to see if it captured any Radiotap header info in the capture file.  The Radiotap header data appears to be absent in the capture file that airodump-ng creates.  I also checked with tshark and got the same result.   There is no data for the wlan_radio.signal_dbm field.  The Radiotap header data doesn't seem to make it into the capture file when I capture with airodump-ng.

2. I also tried using "airodump-ng -r capfile.pcap" and I don't see any numbers for the PWR column.  This is interesting because I can see numbers there when I am capturing, but not when I read from the capture file that airodump-ng produces.

aircrack-ng, 2019-03-19:
1. I know, that's why I asked to capture it with tcpdump or wireshark and check with those tools.
2. And load that capture (made with wireshark/tcpdump that you saved in regular pcap format) with airodump-ng to see if the signal is reported or not.

charliebabbitt1988, 2019-03-20:
I re-tested the capture with tcpdump (tcpdump -i $interface -w tcpdump_test.pcap) and airodump-ng can read that capture file's Radiotap header data, no problem.  I can see data in the PWR column and I can also see data when I open the tcpdump produced capture file in Wireshark.

I also re-tested the capture within the Wireshark GUI and the produced capture file was readable by "airodump-ng -r wireshark_test.pcap".  I can see RSSI data in the PWR column just like the capture file produced by tcpdump.

When I test an airodump-ng produced capture file with "airodump-ng -r airodump-ng_test.pcap", it doesn't show any data for the PWR column.  Similarly, the Radiotap header data can't be seen in airodump-ng_test.pcap when I open it in Wireshark.

Conclusion: for some reason, airodump-ng is not recording the Radiotap header data to its produced capture files.

(I'll try a reboot and re-test with airodump-ng.  I'll report back soon.)

aircrack-ng, 2019-03-20:
Airodump-ng doesn't keep the headers when saving.

This is because it supports multiple headers types and that's in order to simplify parsing it. OSdep takes care of getting the signal/noise information from the frame and passes a structure back to airodump-ng containing that information, along with the packet stripped from its headers. Some headers are hard to parse by some tools and sometimes are buggy, signal is in various parts, and so on. Look into the osdep code and you'll see.

closed 2019-03-20 by aircrack-ng