aircrack-ng/rtl8812au #299: Driver crashes running kernel 5.0

ZerBea, 2019-03-16:
While the driver running fine using kernel 4.19.23-1-lts, it crashes running kernel 5.0.2-arch1-1-ARCH:
4.19.23-1-lts:
$ hcxdumptool -i wlp0s29f7u3
initialization...

start capturing (stop with ctrl+c)
INTERFACE................: wlp0s29f7u3
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries
MAC CLIENT...............: b0ece170192e
MAC ACCESS POINT.........: 00221c066d88 (incremented on every new client)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 64235
ANONCE...................: 5b6c86960af3887e50bf88622cac56620a41d22304db91927863f5541b0be8f6

INFO: cha=11, rx=1266, rx(dropped)=433, tx=79, powned=2, err=0

 5.0.2-arch1-1-ARCH:

$ hcxdumptool -i wlp3s0f0u1 --enable_status=1
initialization...

start capturing (stop with ctrl+c)
INTERFACE................: wlp3s0f0u1
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries
MAC CLIENT...............: b0febdb30d78
MAC ACCESS POINT.........: 544e4585957f (incremented on every new client)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 61451
ANONCE...................: 7f34b17ce8dcd85c287ca7bc39e55d524a490c997243a513e94f0a2eec29d06c

INFO: cha=1, rx=0, rx(dropped)=0, tx=10, powned=0, err=0

Sometimes a few packets are delivered via raw_socket;
INFO: cha=1, rx=233, rx(dropped)=31, tx=99, powned=0, err=0

But mostly no packets are delivered
INFO: cha=1, rx=0, rx(dropped)=0, tx=10, powned=0, err=0

Maybe something went wrong in combination with kernel 5.0, because the driver was working fine running 4.19

Do you see the same behavior running aireplay-ng?

kimocoder, 2019-03-16:
which branch? I've spotted some kernel patches in the wild but haven't really got time to look at it right away.

kimocoder, 2019-03-16:
Take a look https://github.com/lwfinger/rtl8812au/commits/v5.2.6 if you need a quick fix until I'm back

ZerBea, 2019-03-16:
Hi Christian.
Running the default branch, here. Maybe we have some important changes within kernel 5.0. Right now, a noticed that behavior on the mt76x0, too, while mt7601u and rt2800usb are working fine on 5.0.
I'll check https://github.com/lwfinger/rtl8812au/commits/v5.2.6.
BTW:
The rtl8188eu seems to work fine, too (after I noticed that that driver reports the frequency in Hz instead of MHz):
https://github.com/ZerBea/hcxdumptool/issues/48

Cheers
Mike

BTW:

ZerBea, 2019-03-16:
Looks like the driver from here
https://github.com/lwfinger/rtl8812au/commits/v5.2.6
has a regression, too:
Interface wlp3s0f0u1
	ifindex 3
	wdev 0x1
	addr 74:da:38:06:45:e7
	type managed
	wiphy 0
	txpower 12.00 dBm

this driver:
Interface wlp3s0f0u1
	ifindex 5
	wdev 0x200000001
	addr 74:da:38:06:45:e7
	type managed
	wiphy 2
	txpower 18.00 dBm

kimocoder, 2019-03-28:
@kcdtv join me and Alfa Networks om Telegram.
@re4son should also join the party.

Fins me by the Nick

ZerBea, 2019-03-28:
Just tested RTL8187 and noticed that the driver showing the same issues running on kernel >= 5.0.
Even Wireshark doesn't receive packets.
https://github.com/ZerBea/hcxdumptool/issues/42#issuecomment-477551556
That's very interesting, because kernel <= 4.20 working fine.

ZerBea, 2019-03-30:
Running some further going tests and found the cause of the crash. It is related to the radiotap header.
Thanks for your effort.
Cheers
Mike

ZerBea, 2019-04-02:
Closed that issue, because it is possible related to xhci subsystem in combination with USB3.

closed 2019-04-02 by ZerBea