aircrack-ng/rtl8812au #300: No (few) APs detected

jahknem, 2019-03-19:
Running the Kali 2019.1 Rolling Release, but I also tried using Debian Sid as well as Raspberry Sid/Buster.
When using airdump-ng or hcxdumptool I can only get very few access points using the AWUS036ACH (RLT8812AU) or the AWUS1900 (RLT8814AU) compared to AWUS036NHA. This does not only apply to 5 GHz but also 2.4 GHz Networks. I also do not appear to be seeing as many clients as I should. Even if the Adpter is right beside AP and Client it takes multiple minutes to detect them, sometimes it doesn't detect them at all.
Please tell me what files I should provide.

chic01taliano, 2019-03-22:
Same happens to me on my AWUS036ACH running on Kali 2019.1a. Sometimes shows a shitload of APs and sometimes none.

chic01taliano, 2019-03-23:
'airmon-ng check kill' solved the issue for me

huitc, 2019-03-28:
https://github.com/aircrack-ng/rtl8812au/issues/299

kimocoder, 2019-04-07:
Allways do "airmon-ng check kill" to remove interferrence, but now you should also start moving over to the v5.3.4 branch, which hopefully will be a step up :+1:

closed 2019-04-07 by kimocoder