aircrack-ng/rtl8812au #301: Can't compile driver with kernel 5.0.3-200.fc29.x86_64

rickhg12hs, 2019-03-26:
Looks like some kernel changes prevent compilation.

```
$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.0.3-200.fc29.x86_64/build M=/home/rick/rtl8812au/rtl8812au-5.2.20  modules
make[1]: Entering directory '/usr/src/kernels/5.0.3-200.fc29.x86_64'
  CC [M]  /home/rick/rtl8812au/rtl8812au-5.2.20/os_dep/linux/rtw_android.o
/home/rick/rtl8812au/rtl8812au-5.2.20/os_dep/linux/rtw_android.c: In function ‘rtw_android_priv_cmd’:
/home/rick/rtl8812au/rtl8812au-5.2.20/os_dep/linux/rtw_android.c:627:62: error: macro "access_ok" passed 3 arguments, but takes just 2
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
                                                              ^
/home/rick/rtl8812au/rtl8812au-5.2.20/os_dep/linux/rtw_android.c:627:7: error: ‘access_ok’ undeclared (first use in this function)
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
       ^~~~~~~~~
/home/rick/rtl8812au/rtl8812au-5.2.20/os_dep/linux/rtw_android.c:627:7: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:277: /home/rick/rtl8812au/rtl8812au-5.2.20/os_dep/linux/rtw_android.o] Error 1
make[1]: *** [Makefile:1557: _module_/home/rick/rtl8812au/rtl8812au-5.2.20] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.0.3-200.fc29.x86_64'
make: *** [Makefile:1958: modules] Error 2
```
Looking at the kernel sources...
```
$ grep -r 'define access_ok' /usr/src/kernels/
/usr/src/kernels/4.20.16-200.fc29.x86_64/arch/x86/include/asm/uaccess.h:#define access_ok(type, addr, size)\
/usr/src/kernels/4.20.16-200.fc29.x86_64/include/asm-generic/uaccess.h:#define access_ok(type, addr, size) __access_ok((unsigned long)(addr),(size))
/usr/src/kernels/4.20.15-200.fc29.x86_64/arch/x86/include/asm/uaccess.h:#define access_ok(type, addr, size)\
/usr/src/kernels/4.20.15-200.fc29.x86_64/include/asm-generic/uaccess.h:#define access_ok(type, addr, size) __access_ok((unsigned long)(addr),(size))
/usr/src/kernels/5.0.3-200.fc29.x86_64/arch/x86/include/asm/uaccess.h:#define access_ok(addr, size)		\
/usr/src/kernels/5.0.3-200.fc29.x86_64/include/asm-generic/uaccess.h:#define access_ok(addr, size) __access_ok((unsigned long)(addr),(size))
```

rickhg12hs, 2019-03-26:
FYI,  I changed `rtl8812au-5.2.20/os_dep/linux/rtw_android.c` to the following and compilation completed.

```C
#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 0, 0))
        if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
#else
        if (!access_ok(priv_cmd.buf, priv_cmd.total_len)) {
#endif /* Linux kernel < 5.0.0 */
```
There were some compilation warnings for other files but I don't know if they are significant.  I'll do a bit of testing later.

huitc, 2019-03-28:
https://github.com/aircrack-ng/rtl8812au/issues/299

gordboy, 2019-04-01:
This was fixed *ages* ago. You need to grab a more up to date copy of the repo, my friend.

rickhg12hs, 2019-04-01:
> This was fixed _ages_ ago. You need to grab a more up to date copy of the repo, my friend.

I see now that a change was made a couple of months ago.  I was only working with the released ZIP file code and didn't think to look in the git branch with the same version number as the release.  In addition, I didn't see any updates on the master branch, nor did I notice branch v5.3.4.

I now better understand your branch development practices and I look forward to the next release.

Sorry for the noise.

closed 2019-04-01 by rickhg12hs