aircrack-ng/rtl8812au #302: No such file or directory error

huitc, 2019-03-28:
`Linux HuiTC716 4.4.0-17763-Microsoft #379-Microsoft Wed Mar 06 19:16:00 PST 2019 x86_64 GNU/Linux`
`win10 pro` wsl kali

```
$ sudo ./dkms-install.sh
About to run dkms install steps...

Creating symlink /var/lib/dkms/rtl8812au/5.2.20.2/source ->
                 /usr/src/rtl8812au-5.2.20.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.4.0-17763-Microsoft cannot be found at
/lib/modules/4.4.0-17763-Microsoft/build or /lib/modules/4.4.0-17763-Microsoft/source.
Error! echo
Your kernel headers for kernel 4.4.0-17763-Microsoft cannot be found at
/lib/modules/4.4.0-17763-Microsoft/build or /lib/modules/4.4.0-17763-Microsoft/source.
Finished running dkms install steps.
```

```
sudo make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.4.0-17763-Microsoft/build M=/home/huitc/rtl8812au  modules
make[1]: *** /lib/modules/4.4.0-17763-Microsoft/build: No such file or directory.  Stop.
make: *** [Makefile:1562: modules] Error 2
```

aircrack-ng, 2019-03-28:
I doubt any Linux kernel driver will ever work on Windows

tech4him1, 2019-03-28:
@huitc I doubt WSL even has a "Linux kernel", since the whole point of WSL is to enable linux utils to run on Windows, which of course has its own kernel.

huitc, 2019-03-28:
So what should I do? t.t

huitc, 2019-03-28:
```
/usr/lib/modules$ ls
4.19.0-kali3-amd64  4.19.0-kali4-amd64
```

aircrack-ng, 2019-03-28:
Use a Linux VM.

closed 2019-03-28 by aircrack-ng

huitc, 2019-03-29:
really?...😐

aircrack-ng, 2019-03-29:
Or use a Live CD/Linux install.

tech4him1, 2019-03-29:
@huitc If you just need to connect to the wifi, use a Windows driver for the RTL8812AU and WSL should be able to connect as well. If you are wanting to actually use `aircrack-ng` or a tool like that, you likely need to run Linux directly on the hardware (live CD or install).

huitc, 2019-03-29:
```
$ ls
4.19.0-kali3-amd64  4.19.0-kali4-amd64  4.19.0-kali4-cloud-amd64  4.19.0-kali4-rt-amd64  4.4.0-17763-Microsoft
```

huitc, 2019-03-29:
I have installed the wifi adapter driver, but it still can't..