aircrack-ng/rtl8812au #309: It is possible to compile against a moduleless kernel?

noamcore, 2019-03-30:
Hi, I'm actually running a kernel with all modules compiled as builtin, so my question is:

It's possible to compile with the `modules=n` in `.config`?

TheSeven, 2019-05-06:
Yes, it is possible, but might need some minor makefile fixes.
Basically put the git repository into `drivers/net/wireless/rtl8812au` and edit `Kconfig` and `Makefile` in the parent directory to include it.
There is some confusion around RTL8812A vs. RTL8821A config options that might need addressing...

kimocoder, 2019-08-17:
Yes it's possible, but that's a lot of work.. have to rip out the firmware blob and everything..I mean.. it's a LOT of work. still interested? I can point you in the right direction, but it requires some understanding for the language "C" .. let me know

TheSeven, 2019-08-17:
I actually did it successfully on my similarly patched up rtl8812bu driver over here:
https://github.com/TheSeven/rtl88x2bu
It's mashed up with a lot of other changes in the latest commit there, but might still provide some hints

kimocoder, 2019-08-17:
Ok. Here you go.. 

https://github.com/ulli-kroll/rtl8822bu/commit/531a81405f563686a7f4678b39dbe3c0cd566fcb
https://github.com/ulli-kroll/rtl8822bu/commit/ce79bab5d661fb8924c7f79502c54ef116e26265
https://github.com/ulli-kroll/rtl8822bu/commit/fdae222fa171e63c1cd05e6584638526432a5c3d
https://github.com/ulli-kroll/rtl8822bu/commit/37ffe18017ab51b786f93c78a08012c505232c7a
https://github.com/ulli-kroll/rtl8822bu/commit/01dc63dcf3e43edb2a9e4e8ce3d5ffe13a11c779
https://github.com/ulli-kroll/rtl8822bu/commit/bac77c8f7981009b0307ccaad5eebaf9858007ad

kimocoder, 2019-08-17:
I see your using code from this repo :yum: 

https://github.com/TheSeven/rtl88x2bu/blob/0e0788bb53c5066a444dab94fca5ecdb3f87b421/core/rtw_cmd.c#L907

Feel free, of course :yellow_heart: :v:

noamcore, 2019-08-17:
Really thanks @kimocoder, I'll try it out.

closed 2019-08-17 by kimocoder