aircrack-ng/rtl8812au #314: Wi-Fi Direct not advertising.

visbs, 2019-04-05:
Hi,

I am using RTL8812AU based chipset USB Wi-Fi Adapter (Like ASUS AC56 , Linksys WUSB6300) in ubuntu (tried in different OS and Kernel version). 
And I am trying to make use of WI-Fi Direct using wpa_supplicant (P2P Enabled) with user app wpa_cli. 
In case of Intel Wireless Driver, whenever I do P2P_FIND or any scan command, from wpa_cli, my device is visible in any Wi-Fi Direct supported devices like Android, Windows etc.
But in case of Realtek driver (rtl8812au), my device is not visible or listed in Wi-Fi Direct device list, whereas I can get the nearby peers available in my device. As well I am able to connect to any SSID available.
So, am I missing something additional setup required for driver to make my device visible to other peer? 
Any help would be very helpful? 

Thank you,

Hurricos, 2019-04-07:
I have observed this as well - for example, from the v5.3.4 branch, [Miraclecast](https://github.com/albfan/miraclecast) can *see* other P2P-capable devices, but is not actually visible or correctly working for other WiFi-Direct capable devices, using an rtl8811au chipset (with no modifications to the `modprobe.d/88XXau.conf`). For comparison, an rtl8192cu based device worked fine under the same conditions.

Looks like something that might not get fixed due to the focus on stuff that's more important to the aircrack team, like monitor / injection mode -- and understandably so. 

So the driver advertises support for, but does not actually successfully implement

kimocoder, 2019-04-07:
We want everything working! Let's get to work.. start finding the code and diff them and we'll ser. I'll look into it

Hurricos, 2019-04-07:
@kimocoder did you ever test Wi-Fi Direct? I could `git bisect` if you knew a version it worked on, but I'm yet to find one that it works on.

kimocoder, 2019-04-07:
No, i never did

visbs, 2019-04-08:
@kimocoder  @Hurricos  is there any way to solve this issue !!

Hurricos, 2019-04-08:
Not as far as I've seen, unfortunately.

Hurricos, 2019-04-08:
@visbs I have poked through pretty much every version with no luck. You could try messing with the modprobe options (check out os_dep/linux/os_intfs.c).

visbs, 2019-04-08:
@Hurricos  actually I am writing the user application mostly using wpa_supplicant for P2P (WiFi -direct),   
so got minimal ideas in rtl8812 driver architecture.  I am hoping this would be solved by some experts in this category. 
Or is there any way I can connect and rise this issues directly to RTL chipset and driver manufacturer? Once I get the response, I shall post the real path of solution here. 
Thank you,

Hurricos, 2019-04-08:
If you want better debugging tools, use this repository with a rtl8812au-based chipset, alongside `horst`.

The driver manufacturers are far, far behind the diffs here. If you check out [v5.2.20, under documents](https://github.com/aircrack-ng/rtl8812au/tree/v5.2.20/documents), you will find a lot of the release docs that the manufacturers have provided which give you specific instructions for how to get the chipsets to work with *their* drivers ... but good luck escalating it to them.

visbs, 2019-04-08:
@Hurricos  Thank you, I will be trying something from my side too. 
@kimocoder Hoping to get some commits with solution soon.

visbs, 2019-04-09:
@Hurricos @kimocoder  i found the same issue being resolved for 8723bs, 
LINK: https://github.com/TinkerBoard/debian_kernel/issues/16. 
Can we relate something with this?
The issue is related to driver not including probe response.

kimocoder, 2019-04-09:
In the code it actuallly saus.. "FIXME"

[image]

kimocoder, 2019-04-09:
@visbs i just implemented the workaround, could you "git pull" and building/installasjon and reload driver and test?

kimocoder, 2019-04-09:
Good documentation over [here](https://github.com/wawtechnologies/utilities-catchwire-kalitap/tree/master/rtl8811AU_rtl8812AU_linux)

visbs, 2019-04-09:
@kimocoder I tried the workaround, it is still not helping. Did it work for you?

kimocoder, 2019-04-09:
Haven't tried, dont really care about P2P, but with some experience i know and have seen lots of issues om P2P and CONCURRENT mode on Realtek drivers.. dont really have time to investigate that issue right away

Hurricos, 2019-04-09:
@kimocoder Don't worry, this still feels like plenty of support :) 

i've got some C under my belt, so I will pull down and poke around for most of tomorrow.

Hurricos, 2019-04-09:
@visbs If you've gotten the device visible to Android devices, fantastic - it oughta be a logic error somewhere which canbe fixed. I will be able to sit down and look in about 4 hours

visbs, 2019-04-10:
@Hurricos any update ?

visbs, 2019-04-11:
@kimocoder @Hurricos  can you give some instructions on how to debug this driver so that i can dig a little further into the code.
Thanks.

visbs, 2019-04-12:
@Hurricos  can u see rtw_p2p.c there are some todo on p2p.
thank u

Hurricos, 2019-04-15:
Hi, sorry for the delay. Could you bring up the p2p interface with wpa_cli and show me your interaction on the CLI? I ask because I'm not well versed in p2p association and could use some leads.

visbs, 2019-04-16:
http://processors.wiki.ti.com/index.php/WiFi_Direct_Configuration_Scripts
The above link has all the details regarding how wpa_cli commands can be used with respect to WiFi-direct.

Hurricos, 2019-04-16:
Thank you, I will take a look at this when I get a chance.

Fateofiko, 2019-04-19:
Hi everyone!
I think my issue is connected to this one.
I am trying to configure my wifi card (it uses rtl8812au chipset) with wpa_supplicant, just to connect to my home access point! I try to install the original drivers for rtl8812au chipset and i connect with no problem, but with this driver aircrack-ng/rtl8812au just can't connect, of course i tried this in manage mode!
Is somebody tried to make connection to some access point?
I just configure the /etc/wpa_supplicant/wpa_supplicant.conf file, and i know that the configuration is working, because all my wifi dongles, that do not use the rtl8812au driver, are successfully connected!

Hurricos, 2019-04-19:
> Hi everyone!
> I think my issue is connected to this one.
> I am trying to configure my wifi card (it uses rtl8812au chipset) with wpa_supplicant, just to connect to my home access point! I try to [...]

Try using [wicd-curses](https://wiki.archlinux.org/index.php/Wicd).

If you're on a headless machine and that's not an option, please dump the logs from your network management process (`journalctl -e wpa_supplicant`).

Association with an access point was never a problem; this issue is for P2P, not managed, mode.

Hurricos, 2019-04-19:
> original drivers for rtl8812au

Which driver did you try?

Fateofiko, 2019-04-19:
This is the driver that works for me:
https://github.com/gnab/rtl8812au
Amm i am not sure that it is "original" one!!! 

I am not sure how to get the log from the wpa_supplicant:
pi@raspberrypi:~ $ journalctl -e wpa_supplicant
Failed to add match 'wpa_supplicant': Invalid argument

Sorry may be, i have to open another issue, not to spam this one?

Hurricos, 2019-04-19:
> This is the driver that works for me:
> https://github.com/gnab/rtl8812au
> Amm i am not sure that it is "original" one!!!
>
> I am not sure how to get the log from the wpa_supplicant:
> pi@raspberrypi:~ $ [...]

It would be better to open another issue, but it certainly seems like a end-user usage problem.

Fateofiko, 2019-04-19:
> 
Try using wicd-curses.

If you're on a headless machine and that's not an option, 
> 

Yes i use headless raspberry pi 3 model B, may be i should try wicd-curses, and then to ask for help!
So, if association with an access point was never a problem, of course may be i am doing something wrong!
Thanks, i will try wicd-curses!!!

visbs, 2019-04-22:
@Fateofiko  I did try that driver which u specified, which is of no help.

Fateofiko, 2019-04-22:
@visbs This driver works for me, but i am using it for simple connection to my home access point, but you need direct connection, and i am not sure this driver can be useful for that.
I also used this tutorial to build and install the driver:
https://layereight.de/raspberry-pi/2016/08/25/raspbian-rtl8812au.html
Hope, that this will help you  :)!

visbs, 2019-04-23:
@kimocoder just remove the "/" in makefile for uninstall 
from
```
uninstall:
	rm -f $(MODDESTDIR)/$(MODULE_NAME).ko
/sbin/depmod -a ${KVER}
```
to
```
uninstall:
	rm -f $(MODDESTDIR)$(MODULE_NAME).ko
/sbin/depmod -a ${KVER}
```
PS:This has nothing to deal with this issue .

kimocoder, 2019-04-23:
You can do it, so you get the credits too. Thanka

visbs, 2019-04-24:
@kimocoder is this issue solvable !?

visbs, 2019-05-13:
Any progress on this?

kimocoder, 2019-05-25:
@visbs Yes, you should use the newest branch, v5.6.4.
Everything you will need is supplied in the package :1st_place_medal: 

Check it out and report back! Thanks

kimocoder, 2019-08-17:
@visbs 

Hello there people.. in the new branch v5.6.4.1 the WiFi direct is working :1st_place_medal: 
That did take some time, but well well..

[image]

kimocoder, 2019-08-17:
Issue resolved :1st_place_medal:

closed 2019-08-17 by kimocoder

visbs, 2019-08-21:
5.6.4.1 Im not able to build says error: implicit declaration of function ‘rtw_get_wait_hiq_empty_ms’; did you mean ‘rtw_get_time_interval_ms’? [-Werror=implicit-function-declaration]!!

visbs, 2019-08-23:
Okay the driver is now compiling, but pls do look into the concurrent_mode which is not compiling.
Thanks

Azq2, 2022-11-27:
Possible fix for similar driver: https://github.com/jwrdegoede/rtl8189ES_linux/pull/87

But you need backport this changes to this driver

khuongtrinh2, 2024-12-23:
Hi Everyone,
I am working on v5.6.4.1 branch and I have same issue with [visbs](https://github.com/visbs)
I try many solution but it's not successfully and very tired with this issue
Can you share bug-fix about this issue. Thank you so much