aircrack-ng/rtl8812au #315: Unable to change channel using iwconfig

afk11, 2019-04-05:
I need to change the cards channel in software, and with other devices I would run `iwconfig wlan0 channel X`

I've since started using the AWUS036ACH and noticed this doesn't work - it stays at 2.412Ghz. 

I ran `iw dev wlan0 scan` and the channel on the device did change, so it mustn't be impossible. 

I did try `iw dev wlan0 set channel X` and `iw phy phy0 set channel X` and both return `command failed: Operation not supported (-95)`

kimocoder, 2019-04-06:
iwconfig is deprecated.. try using "iw" instead. Have said it before, say it again...

closed 2019-04-06 by kimocoder

afk11, 2019-04-06:
Sorry, did you miss the bottom line?
```
I did try iw dev wlan0 set channel X and iw phy phy0 set channel X and both return command failed: Operation not supported (-95)
```

kimocoder, 2019-04-06:
Ok. Let me Try reproducing it. I've found the IW controls / handlers / implemention code yesterday

MrMas, 2019-05-14:
Can this issue be reopened or should I open a new issue?

afk11, 2019-05-14:
open another, this one was spoiled by having two driver sets installed :+1:

MrMas, 2019-05-14:
Thanks! Sounds good. As it turns out, I've resolved this issue and can set the channel just fine with iw.

kimocoder, 2019-05-14:
Read the README.md. iwconfig is deprecated, stay to "iw"