aircrack-ng/rtl8812au #317: restoring Networkmanager after airmon-ng check kill

GiulianoMalatesta, 2019-04-06:
When I put my wireless adapter Realtek RTL8812AU in monitor mode with the commands

```
airmon-ng check kill; iw dev wlan0 set type monitor

Killing these processes:

  PID Name
 1167 wpa_supplicant
```
Then when I want to restore it in normal mode it won't show anymore in Network manager, even if I try to switch it off with: ip link set wlan0 down and then up. The only way to restore the use of the device with Nm is to unplug the device and re-plug it. Is there a better way to restore the functionality of the device? 
I've also tried to restore the processes with:

```
    airmon-ng stop wlan0mon (or whatever the virtual interface name is)
    service NetworkManager start
    service wpa_supplicant start
```

but unsuccessfully

kimocoder, 2019-04-06:
service network-manager start...

closed 2019-04-07 by kimocoder

GiulianoMalatesta, 2019-04-08:
that doesn't work