aircrack-ng/rtl8812au #318: IW

afk11, 2019-04-06:
Version of #315 in case people only read the title:
https://github.com/aircrack-ng/rtl8812au/issues/315

kimocoder, 2019-04-06:
Allways avoid using iwconfig at least.. that one causes trouble for me trying to put them in monitor til, but IW works again. Leave this open I'll dive into ut 😊

kimocoder, 2019-04-07:
Well, this is how is goes to me, on the same adapter... i may set channels, but scanning is actually off :/
God damn, that needs to be resolved..

kimocoder, 2019-04-07:
I can set channels, but scanning got issues with that chipset.

[image]

kimocoder, 2019-04-07:
There are some information about the missing functions [here](https://forum.openwrt.org/t/solved-rtl8188-driver-for-18-06-01/29131/3l)

bbl!

kimocoder, 2019-04-07:
Utility iw exclusively uses the nl80211 interface.
    some drivers, like RTL8188EU, RTL8188ETV, RTL8188FTV etc. don't implement those ioctls and therefor iw command is not working partially, or at all.

The OpenWrt uses /package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh to manipulate the WiFi and it's based on iw command.
The major issue in it is that it uses simplified way how to do so. First it removes all the interfaces and that creates them again with the new required setup. Generally it works, when the WiFi driver supports the needed commands, but when not, it fails.
I've solved it with the rewriting of iw commands with old-fasshioned ifconfig, iwconfig commands and it works now. I've modified only the AP functions which I need, but it will work for the other functions too.
I will suggest to include simple test if the WiFi driver supports the iw ioctls, or not and than use the appropriate commands in the mac80211.sh.

kimocoder, 2019-04-07:
Just pushed this https://github.com/aircrack-ng/rtl8812au/commit/3a4df6e3dcb753d1e53836da2204680baa3ad478

kimocoder, 2019-04-07:
@afk11 Could you update with "git pull" then "make -j4 && make install" again.. (and of course "modprobe -r 88XXau && modprobe 88XXau" again and test...

afk11, 2019-04-07:
hey @kimocoder! I just did a clean install of my system with v5.2.20, and this time setting channels is working! The previous setup was pretty old and might've had other drivers installed as well, I'm thinking that was my problem now :/

closed 2019-04-07 by afk11

kimocoder, 2019-04-08:
Yeah, thought it was weird..

afk11, 2019-04-08:
Yea, my bad. Thanks for your work on the driver, and for getting back to me also!