aircrack-ng/rtl8812au #319: Wifi adapter disappears after some minutes

matthjes, 2019-04-07:
Hi,
I've installed the package from AUR and while the adapter is first recognized and I can connect to a hotspot, it disappears after some minutes.

Here is the output of dmesg: https://drive.google.com/open?id=1tR9JnPQfRa_adNJy2W-VN9KZJQ2LXiJW

My kernel version is 4.19.32-1.

Please let me know if you need further information.

Hurricos, 2019-04-08:
@matthjes I assume you mean [this AUR](https://aur.archlinux.org/packages/rtl8812au-dkms-git/). This driver doesn't support RTL88XXBU. Check out [here](https://wikidevi.com/wiki/Realtek#ac) for some more info about the differences in chipsets.

kimocoder, 2019-04-08:
Check the docs folder, there is a autosuspend feature that should be disabled.

matthjes, 2019-04-09:
Sorry, this is the wrong driver. I have a RTL8812bu chipset, my mistake.

closed 2019-04-09 by matthjes