aircrack-ng/rtl8812au #320: Explanation needed for disabling IPv6 in dkms-install.sh

diodelass, 2019-04-07:
The `dkms-install.sh` script disables IPv6 system-wide using `sysctl.conf`. This is a rather drastic and unexpected measure for a wireless chipset driver to take (especially in the context of Aircrack-NG, which, to my knowledge, does not usually involve itself directly with IP). Additionally, there appears to be no information about why IPv6 needs to be disabled, either in the script's comments (there are none) or in the `README.md`. Users who make use of IPv6 must notice these lines in the script before they run it (or notice the output it produces and correct the changes after running it), and in any case, they are left with no information about why the installer disables IPv6 or what the consequences of keeping it enabled after installing the driver could be. I don't think this is acceptable. At the very least, there should be a sentence in the `README.md` or a comment in the script file explaining why this action is taken by the `dkms-install.sh` script.

kimocoder, 2019-04-07:
I'll add a switch to it. Setting up my NUC now

kimocoder, 2019-04-07:
I added a switch for it in Makefile. Some boards needs it disabled, simple as that. Turned to "N" by default. Thanks!

closed 2019-04-07 by kimocoder

diodelass, 2019-04-08:
I appreciate the response, but your solution does not directly address the original issue. I am still confused as to why IPv6 needs to be disabled on some systems in the first place, and still have no apparent way of knowing whether I should disable it or not on an arbitrary given system. Nothing in my limited but not insignificant experience with wireless networking leads me to expect that a wireless chipset driver should interfere with IPv6 in any capacity.

The `README.md` provides many other tips and bits of context for the usage of the driver that are far more widely-understood than this (e.g. the reminder to use `airmon-ng check kill`, or the commands to set the interface's `txpower`), so why not add a couple of sentences explaining the problems associated with IPv6, so that users of any of the affected devices know what to do?

If there's something important that I'm not aware of here, let me know, but I'm still confused by the ongoing lack of even a perfunctory explanation for this.