aircrack-ng/rtl8812au #321: Error in make

EmmaDSamms, 2019-04-07:
In file included from ./include/net/ip.h:31:0,
                 from /root/rtl8812au/core/rtw_br_ext.c:19:
./include/net/route.h: In function ‘ip_route_output’:
./include/net/route.h:140:3: warning: missing initializer [-Wmissing-field-initializers]
./include/net/route.h:140:3: warning: (near initialization for ‘fl4.__fl_common.flowic_iif’) [-Wmissing-field-initializers]
./include/net/route.h:141:3: warning: missing initializer [-Wmissing-field-initializers]
./include/net/route.h:141:3: warning: (near initialization for ‘fl4.__fl_common.flowic_scope’) [-Wmissing-field-initializers]
/root/rtl8812au/core/rtw_br_ext.c: At top level:
cc1: warning: unrecognized command line option "-Wno-date-time" [enabled by default]
cc1: warning: unrecognized command line option "-Wno-cast-function-type" [enabled by default]

kimocoder, 2019-04-07:
What system, kernel and device are you compiling on?

kimocoder, 2019-04-07:
Edit the "Makefile" and change these: (add # infront)

#EXTRA_CFLAGS += -Wmissing-field-initializers
#EXTRA_CFLAGS += -Wno-cast-function-type
#EXTRA_CFLAGS += -Wno-date-time

Then do "make clean" and "make -j4" again

closed 2019-04-07 by kimocoder