aircrack-ng/rtl8812au #324: whats the problem??? t.t

huitc, 2019-04-08:
```
root@kali:~/Desktop/rtl8812au# uname -a
Linux kali 4.19.0-kali4-amd64 #1 SMP Debian 4.19.28-2kali1 (2019-03-18) x86_64 GNU/Linux
```
```
root@kali:~/Desktop/rtl8812au# hcxdumptool -I
wlan interfaces:
warning: NetworkManager is running with pid 484
warning: wpa_supplicant is running with pid 864
```
```
root@kali:~/Desktop/rtl8812au# sudo make
sudo make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.19.0-kali4-amd64/build M=/root/Desktop/rtl8812au O="" modules
mmake[1]: ????usr/src/linux-headers-4.19.0-kali4-amd64??
ake install
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: ?????usr/src/linux-headers-4.19.0-kali4-amd64??
root@kali:~/Desktop/rtl8812au# sudo make install
install -p -m 644 88XXau.ko  /lib/modules/4.19.0-kali4-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 4.19.0-kali4-amd64
```
```
root@kali:~/Desktop/rtl8812au# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

huitc, 2019-04-08:
tenda u12

Hurricos, 2019-04-08:
Dump in more output from `dmesg` that occurs when plugging in the dongle.