aircrack-ng/rtl8812au #328: whats the problem?...

huitc, 2019-04-11:
`Operation not supported (-95)`
tenda u12

huitc, 2019-04-11:
```
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 2604:0012  
Bus 001 Device 003: ID 0cf3:7015 Qualcomm Atheros Communications TP-Link TL-WN821N v3 / TL-WN822N v2 802.11n [Atheros AR7010+AR9287]
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

huitc, 2019-04-11:
```
wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether ee:88:2a:bb:6e:38  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

kimocoder, 2019-11-14:
Update to branch v5.6.4.2 and see if that fixes your problem.
If not, feel free to re-open this issue again. Thanks!

closed 2019-11-14 by kimocoder