aircrack-ng/rtl8812au #329: Strong signal reported as  less than -100

solsticedhiver, 2019-04-11:
Using an Alfa Network wifi usb card **ASWUS036AC**, I got RSSI values than are below -100 for strong signal, like AP in the same room or other devices too.

For example,

| Phone | AWUS036AC |
|:-----:|:---------:|
|  -73  |    -38    |
|  -68  |  -56(??)  |
|  -58  |    -108   |

I am using the 5.2.20 branch of the driver

solsticedhiver, 2019-04-13:
That's weird. Did you change something ?

Because right now, still using archlinux (but within a vbox VM this time), I got positive value instead for strong signal, like 10 or 16.

Did you turn the -108 into something like 8

solsticedhiver, 2019-04-27:
Using now both the 5.3 branch and aircrack-ng from git, it appears to be fixed now.

kimocoder, 2019-05-26:
@solsticedhiver That's great! I've got some news for you which uses the 8812au..
There is a more stable and less issues version (v5.6.4) available in our repo, you should check it out!

Thanks a lot for reporting back, and sorry for taking so long time. :1st_place_medal:

closed 2019-05-26 by kimocoder