aircrack-ng/rtl8812au #330: Questions on the adapter AWUS036ACH

LeeTpro, 2019-04-12:
Good afternoon, tell me please, otherwise I’m probably stupid and could not find answers to these questions:
1. Is it possible to raise a wifi access point at the same time in 2.4 and 5 GHz on the adapter AWUS036ACH
2. Is it possible to scan 2 channels 2.4 and 5 GHz or 2 channels at 2.4 and 2.4 GHz in the monitor mode at once? (on the board it seems like this is possible with the datasheet chip)
Thank you in advance

imbarismustafa, 2019-05-06:
Hey. I use my wireless adapter that uses this driver to scan for all networks by specifying channel range with “ -c 1-136 “ in airodump-ng which allows me to display specific channels and different frequencies.

Mister-X-, 2019-05-06:
1. No, you'll need two of them. AP that can do that have two radios
2. No, a radio can only be on ONE channel at ONE time.

closed 2019-11-14 by kimocoder