aircrack-ng/rtl8812au #331: I followed the commands git and CD and sudo. /dkms-install and I restarted but it doesn't work

c0dr7, 2019-04-13:
Please some help, the only thing it said in the end was something about ipv6 I am on Linux mint 19.1 xfce

kimocoder, 2019-04-13:
I fixed the DKMS script.

Use "git pull" and compile/build and reboot to check if it's fixed

c0dr7, 2019-04-17:
hello, after uninstall and reinstall the new one now it doesnt give an error but my dwa171 still doesnt work, I did log out and log in, please give me steps to fix it, thanks for immediate answer!!!

c0dr7, 2019-04-17:
```
Bus 001 Device 009: ID 0bda:1a2b Realtek Semiconductor Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x1a2b 
  bcdDevice            2.00
  iManufacturer           1 D-Link Corporation
  iProduct                2 DISK
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x8a  EP 10 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x0b  EP 11 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)

```

c0dr7, 2019-04-17:
```
/media/c0dr/82F29AA4F29A9C4B/Program Files (x86)/Steam/steamapps/common/dota 2 beta
/media/c0dr/82F29AA4F29A9C4B/Program Files (x86)/Steam/steamapps/common/appmanifest_570.acf
```

c0dr7, 2019-04-17:
```
root@c0dr-cogl:/home/c0dr# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 1c4f:0055 SiGma Micro 
Bus 001 Device 005: ID 04d9:fa59 Holtek Semiconductor, Inc. 
Bus 001 Device 004: ID 045e:078c Microsoft Corp. 
Bus 001 Device 009: ID 0bda:1a2b Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 045e:0083 Microsoft Corp. Basic Optical Mouse
Bus 001 Device 011: ID 2717:ff80  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

c0dr7, 2019-04-17:
which branch should I install?

kimocoder, 2019-05-26:
I'm sorry, but this driver doesn't support "rtl8821cu", only 8812au, 8814au and 8821au.

closed 2019-05-26 by kimocoder