aircrack-ng/rtl8812au #334: [Feature Request] Add support for "TP-Link Archer T2U Nano AC600"

finheuer, 2019-04-18:
I wanted to use the TP-Link Archer T2U Nano AC600 with Parrot OS. I compiled the newest version from branch 5.3.4 and added Product/Vendor ID for the device (2357:011E).

For me there where some issues with network manager but it seemed to work.

Is it possible to add den Product/Vendor ID to the branch so it will be added in newer releases?

I just added
`{USB_DEVICE(0x2357, 0x0122), .driver_info = RTL8812}, /* TP Link - T2U Nano */`
in File
`os_dep/linux/usb_intf.c`

lukaszog, 2019-04-24:
After this modification driver works?

finheuer, 2019-04-25:
At first i thought it would work. But now im running into several erros.

So I wouldn't recommend to push these changes. Sorry for publishing this solution...

I will try to get it running. If i manage to get it running i will reopen this issue with a proper solution.

finheuer, 2019-05-04:
Ok I found my issue after several testing... it was the obvious

I first tried to load it with `.driver_info = RTL8812`, which won't work. But with ` .driver_info = RTL8821` it's perfectly fine.

So with following final code changes it works:
`{USB_DEVICE(0x2357, 0x0122), .driver_info = RTL8821}, /* TP Link - T2U Nano */`

Tested on Parrot OS VM.
 -> WiFi Connection works
 -> Monitor Mode works
 -> Managed Mode works
 -> No errors/warnings in dmes or anything

If you want more testing from me in order to accept these changes you have tell me what I should perform.

asporykhin, 2019-05-06:
> Ok I found my issue after several testing... it was the obvious
> 
> I first tried to load it with `.driver_info = RTL8812`, which won't work. But with ` .driver_info = RTL8821` it's perfectly fine.
> 
> So with following final code changes it works:
> `{USB_DEVICE(0x2357, 0x0122), .driver_info = RTL8821}, /* TP Link - T2U Nano */`
> 
> Tested on Parrot OS VM.
> -> WiFi Connection works
> -> Monitor Mode works
> -> Managed Mode works
> -> No errors/warnings in dmesg or anything
> 
> If you want more testing from me in order to accept these changes you have tell me what I should perform.

Hello finheuer!
I think the correct line should be (0x011E instead of 0x0122):
`{USB_DEVICE(0x2357, 0x011E), .driver_info = RTL8821}, /* TP Link - T2U Nano */`

At least, it works like this for me.

finheuer, 2019-05-07:
Yeah, sure. Copy and pasta always works....

kimocoder, 2019-05-07:
Make a PR to get ut supported or I'll do it when I got time :+1:

xavier83, 2019-05-19:
I have a  Dual Band USB Wifi dongle that works with rtl88x2bu drivers. but none of them seem to support monitor mode issues below. Does this driver by any chance support this device? I am hoping to get monitor mode working on this device.

- https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/issues/9
- https://github.com/its-izhar/rtl88x2bu-driver/issues/1
- https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/2

kimocoder, 2019-05-21:
USB VID/PID added. Support has been added. Thanks!

closed 2019-05-21 by kimocoder

finheuer, 2019-05-23:
> Hello finheuer!
> I think the correct line should be (0x011E instead of 0x0122):
> `{USB_DEVICE(0x2357, 0x011E), .driver_info = RTL8821}, /* TP Link - T2U Nano */`
>
> At least, it works like this for me.

I viewed at the commit and the issue I made in my initial post (Inserting the wrong product id) was unfortunately accepted. So the Product ID should be `0x011E` instead of `0x0122`. Can you fix this? And thank you very much!