aircrack-ng/rtl8812au #335: Asus USB AC68 problem

logan001, 2019-04-20:
I'm on arch (5.0.7) with cinnamon and i've downloaded the 5.3.4 from yesterday (i have no clue how to find out the commit)
The driver almost works. i can see the networks but I can't connect to my network on both 2Ghz or 5Ghz. i can connect with my intel device but not with my asus device
I've tried also wpa_passphrase and wpa_supplicant but no luck:
`sudo wpa_supplicant -i wlp2s0f0u7 -c network.txt 
Successfully initialized wpa_supplicant
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
wlp2s0f0u7: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
`
And that all the output without any connection.
Any suggestion?

logan001, 2019-05-11:
to make an update to the situation. now i'm on 12e3c5d
it seems that the network manager in arch linux does not work with this adapter. Also it does not ignore the adapter even after adding it as unmanaged in NetworkManager.conf.
Disabling NetworkManager service and using wpa_supplicant/dhcpd from terminal seems to connect to router on 5GHz.
iwconfig says 1.3 Gbps but the speed is going up and down between 10 and 5 MBytes.
The intel adapter goes up to 15-16 MBytes on 5GHz.

logan001, 2019-05-20:
seems that nobody care about this issue:
so let me add more info:
the current driver is disaster on Asus USB-AC68 for both softap or connect to router.
1) the new driver can go about up to 100mbps. the old driver goes am to 25-26 MB/s (*8=mbits)
2) the new driver softap even though it shows the same 200Mbps in my android the tv can't even play 1080p movies. forget about 4k movies. and this is the most important feature for me.
so:
i've change 2 files by comparing the new with the old and i was able to compile de old version on kernel 5.0.x.
everything is perfect.
thats why i said the new driver is a disaster. i was curios is somehow driver has an issue with archlinux  but it turns out that it behaves exactly the same on linuxmint where the older driver use to work.
[This](2d0547c99fa029db403fe42192be9907e134b59c) is the commit that works best. probably a few more after it
ive added the check for kernel 5.0 from current one. and now luckily i can compile it on kernel 5. i have no clue if i should add more lines to the old driver and i decided that is better to not risk it and ruin it as the current driver
i don't know what you've done wrong with it but recent builds seems to be way slower that this commit.

kimocoder, 2019-05-26:
@logan001 Sorry, haven't got much time lately.

Can you give me the output from "lsusb", "airmon-ng" and "dmesg | tail" (just after plugging the adapter) please. I'll dive into it :+1:

kimocoder, 2019-06-17:
Aaah, this is the 8814au. Sorry, but the phydm and HAL is old on this chipset, unfortunately.
Sorry for the inconvenience, but the 8811, 8812 and 8821 are working out better atm. 8814au stand still at v5.1.5 still (at least HAL and phydm, which is kinda important for driver matter)

closed 2019-06-17 by kimocoder