aircrack-ng/rtl8812au #339: Cannot build the driver on Raspbian

svenyonson, 2019-05-06:
Using a Pi 3 B+

```
$ uname -a
Linux protomed4 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

$ git clone -b v5.3.4 https://github.com/aircrack-ng/rtl8812au.git
$ cd rtl*
$ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
$ sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile

$ make
make ARCH= CROSS_COMPILE= -C  M=/home/micronets/github/rtl8812au O="" modules
make[1]: *** M=/home/micronets/github/rtl8812au: No such file or directory.  Stop.
Makefile:1711: recipe for target 'modules' failed
make: *** [modules] Error 2
```

satmandu, 2019-05-10:
Note how ```uname -a``` shows that you are running on arm7l which is 32 bit mode.
Thus you need to do this (in the main readme those are the options for the RPI 2/3):
```
$ sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
$ sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
```
You made the changes for the 3B+ assuming you were running an os for the RPI in arm64 mode (such as ARM64 builds of debian or ubuntu.)

Here's what the output of ```uname -a``` looks like on an arm64 OS on the 3B:

```
uname -a
Linux ubuntu3b 5.1.0-050100-generic #201905052130 SMP Mon May 6 02:12:55 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
```

I don't know though if this compiles ok in 32 bit mode for the RPI 3B+. It probably does... but that's for you to find out. :)

kimocoder, 2019-05-26:
@svenyonson I did some updated to both kernel support and ARM RPI (64 also) support.

Can you try updating and build again, and please report back

kimocoder, 2019-11-26:
This issue should be resolved. Please use latest branch.

closed 2019-11-26 by kimocoder