aircrack-ng/rtl8812au #346: How the send Deuth packet under Monitor Mode work

d0ye, 2019-05-16:
I tested the driver and it could perform Deuth packet with aireplay-ng，But another driver(8812AU) I was trying to use couldn't send the packet Since the Basic structure seems the same, So I was wondering any clue How this driver implements this feature (Maybe some Document introduce the driver structure? I'm really not familiar with Driver things), And Maybe I could modify my driver by myself so support this feature, Thx

kimocoder, 2019-05-16:
We support it, don't really have time to explain howto add frame injection, but if you probe through the commits, you will find it though

d0ye, 2019-05-17:
Very Appreciate about the info :)

d0ye, 2019-05-29:
It seems with the same frame injection demo program, Under my 8811 driver with log, It even not calling the **rtw_xmit_entry**  function ,  Instead 8812 will call  **rtw_xmit_entry** and **rtw_monitor_xmit_entry** things, which is a callback registered by **ndo_start_xmit** or other. I'm wondering whether there is a **switch** or **config** affected to this feature so when I run the demo under Monitor Mode  It could call the **rtw_xmit_entry** callback.  Any Insight? @kimocoder

cedricbambooza, 2021-05-11:
pls consider closing the issue, when it's solved by now :)

closed 2021-05-13 by d0ye