aircrack-ng/rtl8812au #347: new usbiids

ZeroChaos-, 2019-05-16:
There may be some overlap here with other bugs (t2u nano, t4u) but I have a small stack of wifi cards which I expect to work with this driver and don't.  How can I help get them supported?

```
[  808.518421] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[  808.518422] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  808.518423] usb 1-1.2: Product: 802.11ac NIC
[  808.518424] usb 1-1.2: Manufacturer: Realtek
[  808.518424] usb 1-1.2: SerialNumber: 123456

[  823.623436] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[  823.623437] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  823.623438] usb 1-1.3: Product: 802.11ac NIC
[  823.623439] usb 1-1.3: Manufacturer: Realtek
[  823.623439] usb 1-1.3: SerialNumber: 123456

[  848.200352] usb 1-1.2: New USB device found, idVendor=2357, idProduct=011e, bcdDevice= 2.00
[  848.200353] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  848.200354] usb 1-1.2: Product: 802.11ac WLAN Adapter 
[  848.200354] usb 1-1.2: Manufacturer: Realtek 
[  848.200355] usb 1-1.2: SerialNumber: 00e04c000001

[  869.447795] usb 1-1.2: New USB device found, idVendor=7392, idProduct=b822, bcdDevice= 2.10
[  869.447796] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  869.447797] usb 1-1.2: Product: AC1200 MU-MIMO USB2.0 Adapter
[  869.447797] usb 1-1.2: Manufacturer: Realtek
[  869.447798] usb 1-1.2: SerialNumber: 123456

[  887.112376] usb 1-1.2: New USB device found, idVendor=20f4, idProduct=808a, bcdDevice= 2.10
[  887.112377] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  887.112378] usb 1-1.2: Product: 802.11ac NIC
[  887.112378] usb 1-1.2: Manufacturer: Realtek
[  887.112379] usb 1-1.2: SerialNumber: 123456

[  900.424477] usb 1-1.2: New USB device found, idVendor=7392, idProduct=c822, bcdDevice= 2.10
[  900.424478] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  900.424479] usb 1-1.2: Product: AC1200 MU-MIMO USB3.0 Adapter
[  900.424480] usb 1-1.2: Manufacturer: Realtek
[  900.424486] usb 1-1.2: SerialNumber: 123456

[  915.528170] usb 1-1.2: New USB device found, idVendor=2357, idProduct=0115, bcdDevice= 2.10
[  915.528172] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  915.528173] usb 1-1.2: Product: 802.11ac NIC
[  915.528173] usb 1-1.2: Manufacturer: Realtek
[  915.528174] usb 1-1.2: SerialNumber: 123456
```

kimocoder, 2019-05-16:
Give med "lsusb" with adaptere. They must be added to "usb_intf.c"

ZeroChaos-, 2019-05-17:
Not really any more info here than was in the above paste, but if it makes you happy:
```
Bus 001 Device 024: ID 2357:0115 TP-Link 
Bus 001 Device 025: ID 20f4:808a TRENDnet 
Bus 001 Device 026: ID 7392:c822 Edimax Technology Co., Ltd 
Bus 001 Device 027: ID 2357:011e TP-Link
Bus 001 Device 030: ID 0bda:c811 Realtek Semiconductor Corp. 
Bus 001 Device 029: ID 0bda:b812 Realtek Semiconductor Corp. 
Bus 001 Device 028: ID 7392:b822 Edimax Technology Co., Ltd 
```

Please note, I looked all these up and they are all 88xxau according to my research, but I don't know specifically which are 12 vs 14 vs 21 or whatever.  I tried adding the ids through the new_id functionality and none of them worked :-(

kimocoder, 2019-05-17:
Easier to read it from lsusb, yes. check them with wikidevi.
Like the on top isn't supported.. it's a rtl8812BU..

https://wikidevi.com/wiki/TP-LINK_Archer_T4U_v3

I can always fork and add support to that driver to, but it's not supported by the rtl8812AU

kimocoder, 2019-05-17:
Same as the "7392:c822" this is another driver unfortunately..

ZeroChaos-, 2019-05-17:
oh, my fault, I missed the distinction when I checked.

the other 5 will be added?  I am happy to test whatever before you commit if you tell me what to test, or I can test after

kimocoder, 2019-05-17:
I think they are rtl8812bu all of them, checked 3 of them, but I for no time to check them all right away

cnlohr, 2019-05-18:
I've bought several dongles off Amazon, and I'm having the issue I just can't find any of the AU dongles.  I'm not sure how to support the RTL8812BU with this driver.  It's not as simple as `CONFIG_RTL8822B=y` since it's not plumbed through.

I can't seem to get packet injection working on the drivers for this chipset, either.  I guess it's just burden you guys have to deal with being the best :(.

kimocoder, 2019-05-24:
@ZeroChaos- take a look at this!!

https://github.com/kimocoder/rtw88-usb

brand new 8822B driver...  and THIS is the case.. 8822B is actually 8812BU supported!
I'll start working on it as I yesterday fixed 8812au v5.3.4 injection and more, and today I finished up some rtl8188eus kernel support + more.

Give me a week and I'll add frame injection and more to it :1st_place_medal:

kimocoder, 2019-05-24:
More information found https://lwn.net/Articles/768372/

kimocoder, 2019-05-24:
People... we've got another branch coming up!
Brace for impact, I can't wait :1st_place_medal: 

[image]

kimocoder, 2019-05-26:
This USB ID's belongs to 8812BU, which is in the 8822B driver unfortunately.
Don't have time to maintain another driver at this time :/

closed 2019-05-26 by kimocoder

kimocoder, 2019-05-26:
@ZeroChaos- you really want the rtw88 driver when it's ready. It uses the newer mac80211 instead of nl80211/cfg80211 mix.

Take a look https://lwn.net/Articles/768372/
It's a driver work-in-progress. Worth following

ZeroChaos-, 2019-05-26:
I have been watching it get added to the kernel for a while now, hoping it
would add usb support for the 8812 :-)

kimocoder, 2019-05-26:
Yeah me too, I'll follow it up. Meeting Realtek and Alfa Networks in Taiwan in august at COSCUP, I'll know what thoughts I'll share with 'em

satmandu, 2020-08-04:
FYI some new consolidation has happened in the rtw88 driver: https://lwn.net/Articles/817736/