aircrack-ng/rtl8812au #348: Paquet Injection not working

bobob15, 2019-05-17:
Hello, 
I have installed the 5.3.4 version of the driver but it is impossible for me to use the packet injection with my Alfa Awus036ACH card (see photo)

I have a friend who bought the Alfa Awus1900 card and it doesn't work either.

Why doesn't it work? 

[image]

Lexus89, 2019-05-17:
Same problem here

kimocoder, 2019-05-17:
which kernel? be more specific. I'll do tests in few days on the matter.

bobob15, 2019-05-17:
I use this Kernel : 4.19.0-kali3-amd64

Lexus89, 2019-05-17:
4.15.0-46-generic #49~16.04.1-Ubuntu

kimocoder, 2019-05-17:
Thanks! I'll do checks om sunday funday. :+1:

cnlohr, 2019-05-18:
Just tested this against my AU card, and I'm having an issue too on `4.15.0-20-generic` (Linux Mint).  I wonder what could have caused such a broad issue.

cnlohr, 2019-05-18:
New data point: None of the functions in `rtl8814au_xmit.c` are called on a new packet xmit if in monitor mode, but in other modes, functions are called as expected.

cnlohr, 2019-05-19:
What commands are you using to enter monitor mode?  It looks like on my system, `iw` puts the system into a mode where injection isn't possible on any drivers.  If I use the rtl8822bu drivers, I can, if and only if I use `iwconfig` get the system into a monitor mode with injection.  Still no dice with rtl8812au.

bobob15, 2019-05-19:
Personally I use "iw" but I have tried other ways like "airmon-ng" and "iwconfig" but the packet injection still doesn't work on rtl8812au (however the monitor mode is well activated)

kimocoder, 2019-05-21:
Ok people!

I've fixed the frame injection on the v5.3.4 branch.
Use "git pull" to update and compile/install again, reload the driver with "rmmod 88XXau" and "modprobe 88XXau" after install. Or simply RESTART the machine to reload.

kimocoder, 2019-05-21:
@bobob15 @cnlohr @Lexus89

bobob15, 2019-05-22:
Nice ! Thanks !

Lexus89, 2019-05-22:
Working here, great job thnx!

kimocoder, 2019-05-22:
Yeah I had to go through some commits that where more or less rushed into the repo.
My bad, but as long as we got it working again.. kernel v5.2 support and other minors will be added when I got more time again

closed 2019-05-22 by bobob15

cnlohr, 2019-05-25:
@kimocoder  Are all corrections also present in the v5.6.4 branch?

kimocoder, 2019-05-25:
No, only 8812au chipset work there, takes some time to rebuild and get HAL to fit in 1 module

cnlohr, 2019-05-25:
Thanks. I will use 5.3.4!

cnlohr, 2019-05-25:
TIL one of the devices I was using is an 8814 and that isn't supported :(

cnlohr, 2019-05-25:
@kimocoder Any idea what I would have to do to get this working on the 5.6.4 branch? (I need 8814 support)

kimocoder, 2019-05-25:
Yes, but it will take some time. I may add a quickfix, give med an hour.
And btw... The 8814au HAL is the oldest, it's from v5.1.5 so IT hasn't all the abilities the other chipsets has. I'll give it a try, but the 8812au is stable and range is good --finally!

cnlohr, 2019-05-25:
Gotcha.  That's OK.  Your work is admirable none the less!  Also, I do have a work around for the RTL8822BU, so it's OK for now.  Thank you for everything you do!

Dema323, 2019-05-28:
@bobob15 does your friend use awus1900 with airgeddon ?

kimocoder, 2019-06-16:
8814au is terrible, just to say it straight forward. It's dirty, buggy and old.. sad but for every release of the "base" (which is from rtl8812au), the 8814au falls even further behind