aircrack-ng/rtl8812au #350: Problem with Airggedon

bobob15, 2019-05-22:
Hello, after having put my card in monitor mode manually, it is impossible for me to choose it in Airgeddon, I have an error (see picture)

Can you help me?
[image]

Dema323, 2019-05-27:
Did you fix it ? Having the same issue

kimocoder, 2019-05-27:
No, they (Airgeddon) needs to add support, just like aircrack-ng (airmon-ng) did, as these drivers does not support virtual interfaces (VIF).

closed 2019-05-27 by kimocoder

bobob15, 2019-05-27:
I found someone who has the same problem on the Airgeddon github but the developer seems to say that it comes from the driver and not from Airgeddon itself

[image]

kimocoder, 2019-05-27:
Like in airmon-ng, it should be possible to add checks if device is in monitor mode and go from there. Else there's nothing to do here until the chipsets get into the mac80211 stack

Dema323, 2019-05-27:
i read the same post thats why i asked if i had the right drivers

Dema323, 2019-05-27:
[https://www.evilsocket.net/2019/02/13/Pwning-WiFi-networks-with-bettercap-and-the-PMKID-client-less-attack/](url) this guy uses the WSU 1900 and uses the drivers on the aircrack git

bobob15, 2019-05-27:
[image]

bobob15, 2019-05-27:
No way to make it work? 
I notice that it worked on version 5.2.20..

bobob15, 2019-05-27:
[image]

Dema323, 2019-05-27:
nope, i got v5.2.20 on my laptop now and it doesn't work
Don't think we need to expect support on this repo

Jo900, 2019-06-30:
v5.2.20 is the only that work properly with airgeddon, only if installed with the realtek-rtl88xxau-dkms
let me know if we can see a fix somewhere..