aircrack-ng/rtl8812au #353: Can't switch to monitor mode

solsticedhiver, 2019-05-23:
Using the driver from v5.3.4 branch (f02ac85), I can't switch my alfa card into monitor mode:

     # airmon-ng start wlp0s11u1
    
    PHY	Interface	Driver		Chipset
    
    phy0	wlp0s11u1	88XXau		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
     (monitor mode enabled)
    Error setting channel: command failed: Device or resource busy (-16)
    Error -16 likely means your card was set back to station mode by something.
    Removing non-monitor wlp0s11u1 interface...
     (monitor mode disabled)

This was working until 12e3c5d at least

kimocoder, 2019-05-23:
Do as the README says. Use "iw"

solsticedhiver, 2019-05-23:
OK. Any explanation ?
Is airmon-ng irrelevant or decommissioned ? or this driver does not work with it and iw is preferred ?

solsticedhiver, 2019-05-28:
Well, kismet (git version) is not able to switch the interface to monitor mode either.

One have to do it manually with **iw**.

Why ? Something is broken , no ? If no, why is it not ?

Hurricos, 2019-05-31:
You must use `iw` to put the device into the right mode because while this device does support monitor mode, it does not appear to support it through a VIF, or virtual interface, due to being based on `cfg80211` rather than `mac80211`. The error message it actually gives is semi-buggy, because it suggests that the software thinks the whole card has been put out of monitor mode - but it's usually that when `airmon-ng` creates a `mon0` VIF, the OS notices and grabs it, thinking it's just another adapter.

Hurricos, 2019-05-31:
You can determine whether you can create a given VIF through (I think) the use of `iw list` - you will see something like:

	software interface modes (can always be added):
		 * AP/VLAN
		 * monitor

if you can create a monitor VIF.

Hurricos, 2019-05-31:
> Why ? Something is broken , no ? If no, why is it not ?

To directly respond to this last bit - yes, something is broken; this driver isn't written using `mac80211`. Given the limited functionality of the hardware interface it presents, it is not all that likely if it will ever be usable in that way.

solsticedhiver, 2019-06-01:
Ok.
So previously that was working. `airmon-ng` turned the card into monitor mode **but** without creating a new virtual interface.
I guess now is that creating the new interface is failing, or there is a message aobut the failing.

One have to distingues 2 things:
- turning monitor mode on
- and creating a virtual interface

The first one works. but it's better to use `ip` ?
Closing this then

closed 2019-06-01 by solsticedhiver

kimocoder, 2019-06-01:
This driver does not support virtual interfaces (aka VIF)

Hurricos, 2019-06-01:
Can you confirm that you can get `airmon-ng` to switch to monitor mode in v5.2.9?

kimocoder, 2019-06-01:
Don't use v5.2.9. it was never really maintained as v5.2.20.2 came right after.
v5.2.20(.2) is "stable", while v5.3.4 works allright, v5.3.4 doesn't like NetworkManager and other interferrence

Hurricos, 2019-06-01:
> Don't use v5.2.9. it was never really maintained as v5.2.20.2 came right after.
> v5.2.20(.2) is "stable", while v5.3.4 works allright, v5.3.4 doesn't like NetworkManager and other interferrence

From the sound of it, then, the OP forgot to `systemctl stop NetworkManager` ...

solsticedhiver, 2019-06-06:
I don't even have NetworkManager installed here...

Hurricos, 2019-06-06:
Be my guest to try and bisect the bug, but there's a natural way around it.