aircrack-ng/rtl8812au #358: NEW v5.6.4 driver

kimocoder, 2019-05-25:
* Needs work to get it working. May take some time, as @fariouche was the one with the skill to make all chipsets in one module

kimocoder, 2019-05-25:
@fariouche do you got some time to help out on the new branch?
it fixes several things, also the LPS, more info here https://patchwork.kernel.org/patch/10928417/

I'll ship you an unreleased device (prototype) for the work :1st_place_medal:

fedux, 2019-05-25:
Hi @kimocoder,  

It looks like you made a mistake and merged the new driver into the v5.3.4 branch. See 1d103b936c39cd1a55d939f12cc8719b1c8f71b8

kimocoder, 2019-05-25:
I know, i know... And I don't have time to clean up the mess right now.
will fix it om some days, anyhow there is no doubt on which is more stable and better.

I'll revert the v5.3.4 branch and continue on the v5.6.4 branch when I'm available again

kimocoder, 2019-05-26:
@fedux Now, I reverted it back again. Thanks for the heads up, sir!

satmandu, 2019-05-31:
At the moment 5.6.4 is a seamless drop-in for 5.3.4 for a USB-AC56 card on ARM64/RPI 3 B.

kimocoder, 2019-06-01:
@dema323 @SubaashNair  I'll explain a bit.. the 8812au and 8821au moves forward, new releases popps up, new HAL, phydm.. while the 8814au (AWUS1900) still uses the v5.1.5 HAL with some modifications to fit our 88XXau collection which original from the 8812au.

Status right now? For 8812au/8821au it's ok, the v5.6.4  will be a good one, but the 8814au still hang behind. At least the HAL and PHYDM is, and they are pretty important in that case.

Realtek ihas startes on a newer mac80211 stack they call "rtw88", perhaps with some negotiation we'll try to convince them to add 8812au, 8821au and 8814au to it, who knows..

Really sad as the device in itself looks pretty awesome with those antennas, right?
Been doing these drivers for some time, and I've learned a lot, but the question I ask myself all the time is.. Realtek... why?? I can't understand why they use chipsets that's poorly supported at all.

kimocoder, 2019-06-17:
[image]

**IW PHY# INFO: OUTPUT**

Wiphy phy8
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
	Band 1:
		Capabilities: 0x1972
			HT20/HT40
			Static SM Power Save
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm) (no IR)
			* 2472 MHz [13] (20.0 dBm) (no IR)
			* 2484 MHz [14] (disabled)
	Band 2:
		Capabilities: 0x1872
			HT20/HT40
			Static SM Power Save
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		VHT Capabilities (0x03c031a2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			short GI (80 MHz)
			TX STBC
			SU Beamformee
			+HTC-VHT
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 867 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 867 Mbps
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5180 MHz [36] (30.0 dBm)
			* 5200 MHz [40] (30.0 dBm)
			* 5220 MHz [44] (30.0 dBm)
			* 5240 MHz [48] (30.0 dBm)
			* 5260 MHz [52] (30.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (30.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (30.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (30.0 dBm) (no IR, radar detection)
			* 5500 MHz [100] (30.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (30.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (30.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (30.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (30.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (30.0 dBm) (no IR, radar detection)
			* 5620 MHz [124] (30.0 dBm) (no IR, radar detection)
			* 5640 MHz [128] (30.0 dBm) (no IR, radar detection)
			* 5660 MHz [132] (30.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (30.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (30.0 dBm) (no IR, radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports scan flush.
	Supported extended features:

closed 2019-08-16 by kimocoder