aircrack-ng/rtl8812au #360: WUS 1900 can't put in monitor mode

Dema323, 2019-05-27:
So i got my drivers from [](https://aur.archlinux.org/packages/rtl88xxau-aircrack-dkms-git)
I can put the card in monitoring mode manually but when i try to use airgeddon or airmon-ng to do it i get an error.
Also when i type `iwconfig wlp0s20f0u2 2> /dev/null | grep Mode | awk ''{print $4} | cut -d ':' -f 2` I get Access instead of monitor. I read this was a drivers issue but it seems i got the right drivers ? Could anyone confirm ?

kimocoder, 2019-05-27:
ip link set <device> down
dev <device> set type monitor

Dema323, 2019-05-27:
[https://www.amazon.de/Network-AWUS1900-802-11ac-Ultra-adapter/dp/B01MZD7Z76/ref=sr_1_1?crid=6I87PIU8ZD21&keywords=alfa+awus1900&qid=1558960443&s=gateway&sprefix=alfa+aw%2Caps%2C140&sr=8-1-spell](url)

drivers  :[https://aur.archlinux.org/packages/rtl88xxau-aircrack-dkms-git](url)
Manjaro kernel 4.19

OscarAkaElvis, 2019-05-28:
Yes, to work with the standards is needed to set "Managed" or "Monitor"... is not valid "Access" or "Auto" or that kind of stuff. It seems that is causing a lot of problems for the compatibiliby with some different softwares.

Hurricos, 2019-05-31:
This is a deeply ingrained 'bug' which would more or less require a migration to `mac80211` to fix. Very few modern non-`mac80211` drivers support monitor mode.

kimocoder, 2019-06-16:
@Hurricos Correct.. that would have solved most of our issues. More capabilities, more stable, more support, less bugs, virtual interface (VIF) support... yeah you name it. Fiddling around with old nl80211/cfg80211 dirty I/O mix from back in the days.

Hopefully, Realtek is working on "rtw88", just cross some fingers and toes that they'll add 88xx support (not only 8822b(x) as planned.

Current status is they're finishing up PCI bus & SDIO before they start adding USB.

**Sources / Information below**
https://lwn.net/Articles/768372/
https://patchwork.kernel.org/cover/10624625/
https://www.phoronix.com/scan.php?page=news_item&px=Realtek-RTW88-WiFi-Linux-Driver
https://cateee.net/lkddb/web-lkddb/RTW88_8822BE.html

patchMonkey156, 2019-06-29:
First, disable NetworkManager, even if temporarily - it interferes with setting the device in Monitor Mode.

Try the 5.3.4 branch of this driver for the AWUS1900AC - it uses the chipset RTL8814au - the master, 5.6.4 and 5.6.4.1 branches of this driver do not support *our* wifi card. 

There is an outstanding pull request to add this support - you could try that version if you like: https://github.com/aircrack-ng/rtl8812au/pull/386 - I'm sure that author would appreciate the independent testing.

If that doesn't work, the parent driver (stale as it may be) should still be usable. The "Master" branch was never intended to be used, only for heavy development. v4.3.21 works well for x86/x86_64, v5.1.5 is better for ARM. 

If/when you build the astsam driver, remember - the build command is:
make RTL8814=1
Otherwise, you'll build a driver for RTL8812au. Read the README.md on the webpage for the branch.

At the rest of you:

ieee80211 drivers (I'll personally vouch for the astsam/rtl8812au ieee80211 driver this one is forked from, just not the master branch - the author used that for incomplete development) seem capable of even VIF. The author wrote that driver specifically to fit the Kali Linux wifi penetration testing course which uses that feature. This course covers (and then needed) a modern device using such an ieee80211 driver. Even astsam's version is forked from that original kali-centric driver (which has since been deleted), which itself was an 8812-centric reimplementation of RTLwifi 's original "skeleton". Several concurrent versions popped up using this approach. 

I understand it'd be nice to have an "official" realtek driver, but they likely aren't going to be including the features we come to expect from aircrack_ng. Sure, someone can patch them in... but are they really going to do that?

but here we are with several (hopefully working) implementations. I step away for a few months, and you make it sound like it all went to pot... I guess I'll need to roadtest aircrack-ng/88XXau against https://github.com/aircrack-ng/rtl8812au/pull/386astsam/RTL8812au

Either way, all of them are kinda messy - its all built on a skeleton of RtlWifi implemented for 8187. I would have expected any references to the old chipsets to have been renamed by now - but no, in the Makefile RTL8814's phydm details are still found using a reference for a much older chipset. What is aircrack doing differently? And what do the different branches around here even *do?*

kimocoder, 2019-11-11:
Everyone should move to v5.6.4.2 as Realtek released the new base and I've added lots of patches on top. Other branches will get left behind as the focus is on the latest, at all times.

The issue is resolved also btw, but if any other issues is encountered, feel free to open new issue reports on each subject. Thanks!

8814AU on version v5.6.4.2 with kernel v5.3.7 below.

[image]

closed 2019-11-11 by kimocoder

patchMonkey156, 2019-11-11:
Any suggestions for how to patxh this into the kernel sources directly,
without dkms? A little bit of a specialized use case, I know. But useful
none the less.

kimocoder, 2019-11-11:
Not really, found a Android kernel patch not long ago thus for the v5.3.4

https://raw.githubusercontent.com/kimocoder/DJY_88XXAU_For_4.4__4.9_4.14_4.19_Android_Kernel_Patch/master/DJY_88XXAU_For_4.4__4.9_4.14_4.19_Android_kernel_.patch