aircrack-ng/rtl8812au #361: Supporting RTL8811CU Chipset [Comfast 600Mbps)

breisig, 2019-05-28:
I have a Rtl8811cu Comfast 600Mbps device [Bus 001 Device 002: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC] and it would be great if it would be possible to get it supported by this driver?

Hurricos, 2019-05-28:
> I have a Rtl8811cu Comfast 600Mbps device [Bus 001 Device 002: ID
> 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC] and it would be
> great if it would be possible to get it supported by this driver?

The chipsets are pretty different, and the maintainers just received a new dump of v5.6.4. This driver already includes a variety of different chipsets, and that multi-chipset functionality was just broken by the new dump and waiting I think on @fariouche to round back and fix.

I'm hardly an authority on this, but it does not appear that this is going to be possible. Please check out the one of the drivers from [this rtl8821CU](https://github.com/whitebatman2/rtl8821CU/network) network. I think whitebatman2's works well (VHT STA and AP), though it doesn't support monitor mode.

kimocoder, 2019-05-28:
@Hurricos could not answer that better myself

breisig, 2019-05-29:
Thank you for the links.

closed 2019-05-29 by breisig

karlkwan, 2020-04-29:
Realtek RTL8811CU/RTL8821CU USB wifi adapter driver version 5.4.1 for Linux 4.4.x up to 5.6.x
-> (https://github.com/brektrou/rtl8821CU))