aircrack-ng/rtl8812au #368: Is this driver suitable for TP-LINK Archer T2U Nano?

niXman, 2019-06-03:
Hi,

I'm trying to use this driver for  TP-LINK Archer T2U Nano. After installing the driver, I still do not see the `wlan0` network interface.

Ideas?

kimocoder, 2019-06-03:
lsusb & airmon-ng output thanks

niXman, 2019-06-03:
lsusb output:
```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 04d9:1702 Holtek Semiconductor, Inc. Keyboard LKS02
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0eef:0005 D-WAV Scientific Co., Ltd 
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 003: ID 2357:011e  <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

> airmon-ng

```
PHY	Interface	Driver		Chipset
```

dmesg when I plug the device:
```
[ 7680.896257] usb 5-1: new high-speed USB device number 3 using ehci-platform
[ 7681.031274] usb 5-1: New USB device found, idVendor=2357, idProduct=011e
[ 7681.031308] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7681.031329] usb 5-1: Product: 802.11ac WLAN Adapter 
[ 7681.031349] usb 5-1: Manufacturer: Realtek 
[ 7681.031369] usb 5-1: SerialNumber: 00e04c000001
```

niXman, 2019-06-03:
I added the ID of my device here(https://github.com/aircrack-ng/rtl8812au/blob/v5.3.4/os_dep/linux/usb_intf.c#L221) and now I can see the `wlx503eaae7d39e` interface.

Does anyone have any idea how I can switch this device to 5.8 Ghz?

niXman, 2019-06-03:
Also I was able to create a Access Point, but I need to switch it to 5.8

3ternalDeath, 2019-06-04:
From [Issue #335 ](url)
The wrong usb code was provided for this device

niXman, 2019-06-04:
@3ternalDeath Why do you think so? How to correctly get the code?

3ternalDeath, 2019-06-04:
Disregard, looks like you fixed it locally after cloning

niXman, 2019-06-04:
Ah, understood.

Maybe someone knows how I can find out if this device supports the creation of an AP for 5.8?

kimocoder, 2019-06-04:
NO_IR is from the wifi_regd.c (the regdom)

kimocoder, 2019-06-04:
Make a PR with the correct VID/PID and I'll add it. Best you do it, so credits goes your way

niXman, 2019-06-04:
@kimocoder I was see the `wifi_regd.c` file but I dont understand what should I add?

> correct VID/PID

why you think `2357:011e` is not correct?

niXman, 2019-06-04:
another question is why the USB ID `2357:0122` marked as `TP Link - T2U Nano`? Is it normal that the same device has two identifiers because my `T2U Nano` has USB ID `2357:011e`?

niXman, 2019-06-05:
can anyone help me?

sedlund, 2019-06-16:
I also have a TP-Link T2U Nano shows as 2357:011e

kimocoder, 2019-06-16:
I'll add it in some hours, which branch are you using?

niXman, 2019-06-16:
@kimocoder `v5.3.4`

kimocoder, 2019-06-16:
Gone through the VID/PID's and added what where missing and "vice versa" (on branches)

Happy go fishing :+1:

closed 2019-06-16 by kimocoder

niXman, 2019-06-17:
@kimocoder Hi,
about this commit: https://github.com/aircrack-ng/rtl8812au/commit/e3281319f663305c3ccc4d284ec379cbfec2e92b

I did it myself, and I wrote about this above. But my question was, what do I need to do to switch to 5.8(I wrote about this above also)? You told me `NO_IR is from the wifi_regd.c (the regdom)` but I still do not understand what should I do?