aircrack-ng/rtl8812au #369: The Driver Causes a 80 ms Delay for ALFA AWUS 036ACH

researchyao001, 2019-06-06:
Hello,

I installed this driver (rtl8812au, version 5.2.20) for my USB WiFi adapter, ALFA AWUS 036ACH which uses chipset rtl8812au. I installed the driver for Ubuntu 16.04. The kernel version is 4.15.0. With the driver, my USB adapter can send and receive packets. However, the driver causes a 80 ms delay for packets sent. After I do a calculation, it can tell there is a buffer of about 200k bytes in the driver. But I didn’t find it. Can you please tell me where is the buffer in the driver or in the kernel space? How Can I reduce the size of the buffer?

kimocoder, 2019-06-16:
@researchyao001 If you got the 8812au chipset (as the 036ACH has), you could move on to either v5.3.4, but even better the v5.6.4 (even though it isn't quite "ready" yet, but it works great)

kimocoder, 2019-11-14:
@researchyao001 

Update to v5.6.4.2, that would be a big jump and the problem should be resolved.
However, if it's not resolved.. feel free to re-open this issue

closed 2019-11-14 by kimocoder