aircrack-ng/rtl8812au #372: Compile Error

Kradziej, 2019-06-07:
I get this when I try to build under armbian

/root/rtl8812au-aircrack/os_dep/linux/os_intfs.c: In function ‘netdev_vir_if_close’:
/root/rtl8812au-aircrack/os_dep/linux/os_intfs.c:2651:2: error: ‘wdev’ undeclared (first use in this function)
  wdev->iftype = NL80211_IFTYPE_MONITOR;
  ^~~~

So code is:
 
#ifdef CONFIG_IOCTL_CFG80211
       wdev->iftype = NL80211_IFTYPE_MONITOR;
       wdev->current_bss = NULL;
	
shouldn't it be like

#ifdef CONFIG_IOCTL_CFG80211
       padapter->rtw_wdev->iftype = NL80211_IFTYPE_MONITOR;
       padapter->rtw_wdev->current_bss = NULL;

or maybe it's not needed at all?

kimocoder, 2019-08-17:
@Kradziej Sorry the delayed answer.. first off.. which branch are you using? what device/chipset do you have? "lsusb" and "airmon-ng" output would be great.

kimocoder, 2019-11-26:
This issue should be resolved, please use latest branch

closed 2019-11-26 by kimocoder

webrouse, 2019-12-05:
@pdp7 

Hi, I have same problem here:
https://github.com/aircrack-ng/rtl8812au/blob/1af7edca86138533d04d7b7eb08a883d9ea4872a/os_dep/linux/os_intfs.c#L2804-L2810

```
/home/user/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-sunxi_cortexa53/rtl8812au-ct-custom-1af7edca/os_dep/linux/os_intfs.c:2805:2: error: 'wdev' undeclared (first use in this function); did you mean 'cdev'?
  wdev->iftype = NL80211_IFTYPE_MONITOR;
  ^~~~
  cdev
```

LinkerOK, 2021-05-22:
Use the latest branch 5.7.0

LinkerOK, 2021-05-22:
> 
> 
> Use the latest branch 5.7.0

git clone -b v5.7.0 https://github.com/aircrack-ng/rtl8812au.git