aircrack-ng/rtl8812au #374: Freeze when performing sudo iw dev wlan1 del

marc-y-marc, 2019-06-11:
Branch: 5.3.4 (also tested the newer 5.6.x driver branch)
OS: Tested on Raspbian & Ubuntu

**Steps to reproduce:**

- Install driver on clean Raspbian or Ubuntu (18.04.2) use make install or dkpg (both tested)
- sudo iw dev wlan1 del

sudo / iw process will freeze. Only operation 'pull the plug' makes the raspberry pi3b+ live again.  Killing the process will not work, 

I've also tried using airmon-ng start / stop. That gave me the following:

> You already have a wlan1 device but it is NOT in station mode.
> Whatever you did, don't do it again.
> Please run "iw wlan1 del" before attempting to continue

marc-y-marc, 2019-06-13:
Related issue (which was closed but not solved): https://github.com/aircrack-ng/rtl8812au/issues/227

kimocoder, 2019-11-26:
Please use latest branch more less issues.

closed 2019-11-26 by kimocoder