aircrack-ng/rtl8812au #377: Install on Raspbian

veganinside, 2019-06-16:
Hi, seems that is not working the sudo make on raspbian, is only working for Arch?

kimocoder, 2019-06-16:
Mainly for Kali, but should be working on other platforms.
which branch? which RPI board?

Got more hardware incoming (RPI's), so we'll fix it

veganinside, 2019-06-16:
Raspberry B+. Running raspbian little. So Kali would be fully working?

kimocoder, 2019-06-16:
https://gitlab.com/kalilinux/build-scripts/kali-arm

veganinside, 2019-06-17:
Is any branch working good on Raspbian? any similar experiences?
Thanks.

juliogonzalez, 2019-06-17:
Yes, master works on a Raspberry Pi 3B+, Raspbian 9.9 and kernel 4.19.42-v7+  for me with with the instructions at the `README.md`, plus a small tweak to `dkms-install.sh`:
```
diff --git a/dkms-install.sh b/dkms-install.sh
index 547a6f4..1af64d8 100755
--- a/dkms-install.sh
+++ b/dkms-install.sh
@@ -14,7 +14,7 @@ DRV_VERSION=5.3.4
 cp -r ${DRV_DIR} /usr/src/${DRV_NAME}-${DRV_VERSION}
 
 dkms add -m ${DRV_NAME} -v ${DRV_VERSION}
-dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
+ARCH=arm dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
 dkms install -m ${DRV_NAME} -v ${DRV_VERSION}
 RESULT=$?
```
Otherwise the `ARCH` is set to armv7l if you enable `CONFIG_PLATFORM_ARM_RPI` (and no rules are found to build for this architecture) or `ARCH` is set to arm64 if you enable `CONFIG_PLATFORM_ARM64_RPI` and then:
```
DKMS make.log for rtl8812au-5.3.4 for kernel 4.19.42-v7+ (armv7l)
lun jun 17 23:11:07 CEST 2019
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.42-v7+/build M=/var/lib/dkms/rtl8812au/5.3.4/build O="" modules
make[1]: se entra en el directorio '/usr/src/linux-headers-4.19.42-v7+'
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/5.3.4/build/core/rtw_io.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
```
I suspect this is happening because while the CPU is arm64, the system and gcc are not (Raspbian still didn't make images or packages, AFAIK).

@kimocoder maybe a note at the `README.md` file could help other users?

kimocoder, 2019-11-11:
This issue has been resolved in the latest v5.6.4.2. Other branches will be left behind.
Anyhow, if the problem isn't resolved for somehow, or you got other issues, please re-open or create a new one for each specific. Thanks!

closed 2019-11-11 by kimocoder