aircrack-ng/rtl8812au #380: AWUS036ACH can't put it in monitor mode.

kaspertje100, 2019-06-20:
Hi,

I got Linux mint 19.1 installed (4.15.0-52-generic)

With DKMS:
rtl8812au, 5.3.4, 4.15.0-52-generic, x86_64: installed

Wi-Fi adapter: Alfa AWUS036ACH.

adapter does show up with iwconfig:
wlx00c0caa65183  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

And is usable in managed mode. However, when I try to enable monitor mode by either:
sudo ifconfig wlx00c0caa65183 down
sudo iwconfig wlx00c0caa65183 mode monitor
sudo ifconfig wlx00c0caa65183 up
I get: 
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlx00c0caa65183 ; Invalid argument.

or:
sudo iw dev wlx00c0caa65183 set  monitor none
command failed: Operation not supported (-95)

 Any suggestion on how to enable monitor mode for this device?

kaspertje100, 2019-06-20:
Little update: so I got the driver to work.
I had to unload original driver: sudo rmmod 8812XXau and load modprobe 8812au

kimocoder, 2019-06-20:
Yeah, anyhow.. 8812au users could use v5.6.4, better range, more stable driver.
It doesn't support 8821au and 8814au yet, bit 8812au.. ooh yes

ZerBea, 2019-06-24:
@kimocoder:
Default branch is affected by kernel-issue in xhci system (not fixed, yet):
[ 5516.769643] usbcore: registered new interface driver rtl88xxau
[ 5521.897523] usb 1-10.4: new high-speed USB device number 5 using xhci_hcd
[ 5522.027277] usb 1-10.4: New USB device found, idVendor=7392, idProduct=a812, bcdDevice= 2.00
[ 5522.027281] usb 1-10.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5522.027283] usb 1-10.4: Product: Edimax AC600 USB
[ 5522.027286] usb 1-10.4: Manufacturer: Realtek
...
[ 6168.802992] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
Please read more here:
https://bugzilla.kernel.org/show_bug.cgi?id=202541

v5.6.4 (branch) doesn't detect the device (ID 7392:a812 Edimax Technology Co., Ltd):
[ 6766.388560] usb 1-10.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6766.388562] usb 1-10.4: Product: Edimax AC600 USB
[ 6766.388564] usb 1-10.4: Manufacturer: Realtek
[ 6766.388566] usb 1-10.4: SerialNumber: 00e04c000001

kimocoder, 2019-11-11:
@kaspertje100 

Users should move over to v5.6.4.2 branch, all the others is going to hang behind and be unsupported or buggy. Monitor mode is working very well too. Closing issue

[image]

[image]

closed 2019-11-11 by kimocoder

irkenempire, 2020-04-08:
do you have any idea this happend to me too

https://www.youtube.com/watch?v=FAhTpzc7WT0